LumaTeam/Luma3DS #505: InputRedirections stops when exiting System Settings

HoLLy-HaCKeR, 2017-06-05:
**System model:**

o3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (X)

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

When exiting System Settings with Input Redirection enabled, Input Redirection will be shut off. It can be turned on again without problems.

Aditionally, opening the Rosalina menu inside System Settings crashes the 3DS, making the display corrupted.

**Steps to reproduce:**

1. Boot up 3DS with Luma 8.0

2. Start Input Redirection from the Rosalina menu

3. Enter and then exit System Settings

**Dump file:**

/

FrozenDragoon, 2017-06-05:
This is likely due to the fact that exiting system settings counts as a reboot. So, not a bug, but little Luma can do about it. 

From the release notes: "This takes effect until the next boot".

HoLLy-HaCKeR, 2017-06-05:
Reboot using short press on power button does not do this though.

TuxSH, 2017-06-05:
The console reboots on System Settings exit, same thing with Pokémon S/M on O3DS, etc.

closed 2017-06-05 by TuxSH