LumaTeam/Luma3DS #506: Starting Input Redirection before Home Menu Titles are loaded stops them from loading

HoLLy-HaCKeR, 2017-06-05:
**System model:**

o3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (X)

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Starting Input Redirection before all titles/badges on the home menu are loaded, will stop them from loading at all. Input Redirection will also fail with error code 0x09401bfe after a ~5 seconds. Loaded icons did not seem to launch for me.

**Steps to reproduce:**

1. Start 3DS

2. After the Home Menu appears, but before all icons are fully loaded, open the Rosalina menu and start Input Redirection. (will most likely fail)

3. Exit Rosalina and optionally try to launch a title.

**Dump file:**

/

HoLLy-HaCKeR, 2017-06-05:
Might be related to #500

TuxSH, 2017-06-05:
Not related, but I don't think I can fix it. Wait for the menu to be fully started.

closed 2017-06-05 by TuxSH