LumaTeam/Luma3DS #507: System Settings Crash

adyngy, 2017-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

After updating to boot9strap 1.2 and Luma3DS 8.0 , going into the System Settings crashes the 3DS.
 Processor: ARM11 (core 3)
Exception type: data abort
Current process: qtm (0004013000004202)

[ Picture of crash](http://i.imgur.com/MQsdSC1.jpg)

**Steps to reproduce:**

1. Go into System Settings

2. Experience crash

**Dump file:**

[crash_dump_00000006.zip](https://github.com/AuroraWright/Luma3DS/files/1051724/crash_dump_00000006.zip)

shinji257, 2017-06-05:
Duplicate: https://github.com/AuroraWright/Luma3DS/issues/501

HoLLy-HaCKeR, 2017-06-05:
This is not a duplicate, this issue mentions a crash when opening the Settings app, not when exiting the NDS Wifi settings.

fincs, 2017-06-05:
Cannot reproduce here. European New 3DS here.

shinji257, 2017-06-05:
In that case I can't reproduce it either.  USA New 3DS (not the XL version).

HoLLy-HaCKeR, 2017-06-05:
For sake of completeness, can't reproduce on o3DS 11.4.0-37E

TuxSH, 2017-06-05:
Closing, then.

Faulty face-tracking hardware maybe?

closed 2017-06-05 by TuxSH

VKGeovani, 2020-10-21:
bonjour 
je suis nouveau j'aimerais savoir si vous pouvez m'aider, j'ai acheter un 3ds old a mon fils et il a égaré la carte sd j'ai acheté une nouvel carte sd et copier les donner qui se trouvais sur ma 3ds du cout quand j'allume sa 3ds j'ai une erreur et plusieurs code qui sa fiche en conclusion  j'ai se code en particulier   (( sd:/luma/dump/arm11/crash_dump_00000001.dmp))   j'ai besoin de votre aide svp merci