LumaTeam/Luma3DS #508: Update to latest stable - Unable to power on

alzeric, 2017-06-05:
**System model:**

N3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.3.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**

SigHax

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

7.1 stable - updated to Latest Stable 8.0

**Explanation of the issue:**
Updated with LumaUpdater to latest stable 8.0 tonight ... it powered off after successful update.  Now when I try powering on the power light turns on briefly (maybe 500ms) blue light then immediately turns off.  Replaced battery and still same issue even while on charging cable.  Was working flawlessly before update...any assistance would be greatly appreciated.

**Steps to reproduce:**

1.simply updated to latest stable through lumaupdater

2.

**Dump file:**

Can't power on to even do a dump

HoLLy-HaCKeR, 2017-06-05:
Do not use LumaUpdater, instead follow Plailect's guide for [Updating B9S](https://3ds.guide/updating-b9s). This includes Luma 8.0.

HatchlingByHeart, 2017-06-05:
Put your MicroSD card in a computer and restore the backup made by LumaUpdater:

- > Delete "boot.firm"
- > Rename "boot.firm.bak" to "boot.firm"

alzeric, 2017-06-05:
@HatchlingByHeart that worked thank you so much... Had this about a week and already breaking stuff lol.

@HoLLy-HaCKeR I'll give that a read through again for next time, I followed that guide step by step to load all this in the first place I should have know because I actually read that before I homebrewed this in the first place.

closed 2017-06-05 by alzeric

HoLLy-HaCKeR, 2017-06-05:
The article/page I mentioned is specifically for people on Luma3DS 7.1, since B9S had a breaking change and Luma3DS 8.0+ relies on it. It is not the page for installing B9S on a clean system. I recommend following the [/r/3dshacks subreddit](//www.reddit.com/r/3dshacks/) for updates on 3DS homebrew so you know about such things.