LumaTeam/Luma3DS #515: Input redirector disabling 3ds joystick

Jan200101, 2017-06-05:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E Sysnand

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0-21db0d4

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (Before)

PIN lock: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

 3ds Joystick stops responding when Input redirector is active and ingame (tested with Pokemon Omega Ruby and Smash Brothers)

In pokemon it happend right after launching the save

In smash it happend after starting a online game

**Steps to reproduce:**

1. Start Input Redirector from Rosalina

2. Start a game

3. Try using the 3ds joystick ingame

sora10pls, 2017-06-05:
#487 

>The Rosalina menu reads buttons itself in the most basic way possible (through reading the HID_PAD hardware register). For the same reason InputRedirection won't work for it either.

Jan200101, 2017-06-05:
Im talking about the normal joystick not the c-stick

TuxSH, 2017-06-05:
Make sure that your remote joystick is correctly calibrated, also start inputredir outside of the game.

For ΩRαS, while it'll prevent the game from disconnecting from the Internet, it will unfortunately prevent the game from using Internet features itself.

closed 2017-06-05 by TuxSH

Jan200101, 2017-06-06:
I am using a keyboard and the remote Joystick is fine.

Its the 3ds joystick that is causing the issue... You know
this thing.
[image]

After more trying I cant get it to work after I connect my Input Redirection client to it, not even on the home menu