LumaTeam/Luma3DS #516: Opening the Rosalina while playing Pokemon Sun crashes both displays

TheMCNerd2017, 2017-06-05:
Pretty self-explanatory issue, opening up the Rosalina Menu while playing Pokemon Sun(in the overworld, battles, cutscenes, etc) causes both screens to crash with corrupted graphics. The system itself does not crash, merely the screens since everything else is still functional. This issue persists until I power-off the console.

Here is a link to the image I took of the screen crash: http://www.imgur.com/a/YJ8Fx

Gudf, 2017-06-05:
Have you tried it several times ? It crashed the first time for me, but everything is working just fine now, I can open and close the Rosalina Menu, nothing crashes.

TheMCNerd2017, 2017-06-05:
Yes, I tried it several times. Every single time the display would crash, but the game and all other things would still be fully functional.

sora10pls, 2017-06-05:
From release notes:

>Minor issues: the menu can sometime freeze in some games (that will be attempted to be fixed before next release)

TheMCNerd2017, 2017-06-05:
I read the release notes, so I know that. Just thought that I would report the issue as well as the game that causes the issue.

bladesdnm, 2017-06-05:
can we just have an option to disable the menu combo from opening it.

Gudf, 2017-06-05:
I may have found a fix, but I'm not sure. Try checking or unchecking "Enable Game patching", launch the game, launch Rosalina Menu, close it and it should work (you can check or uncheck again "Enable game patching" after this).

Margen67, 2017-06-05:
@bladesdnm #496

TuxSH, 2017-06-06:
This happens somtimes. I'll attempt to fix that later (if possible) after fixing other issues.

> can we have an option to disable the button combo from opening Rosaline menu please.

Set it to an impossible combo.

closed 2017-06-06 by TuxSH

bladesdnm, 2017-06-06:
why even have the option to set combo if it don't save