LumaTeam/Luma3DS #518: NTR CFW crashes sometimes

IngeniousDefault, 2017-06-05:
New Nintendo 3DS

FW: 11.4E

Exploit: boot9strap 1.2

CFW: Luma 8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

Hi there.

After updating to Luma 8.0 after launching the BootNTR Selector 2.8 and booting NTR CFW 3.5 I sometimes get an ARM11 error and the system crashes.

**Steps to reproduce:**

1. Launching BootNTR Selector

2. Launching NTR CFW 3.5

**Dump file:**
[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1052176/arm11.zip)

HoLLy-HaCKeR, 2017-06-05:
This is probably an error on NTR's end, and the only reason you're seeing the error message is because it is enabled by default in Luma 8.0

Jan200101, 2017-06-05:
I have tried NTR 3.5 and it seems to be working fine on my end.
As HoLLy mentioned its probably just a issue on NTR's end

IngeniousDefault, 2017-06-05:
Well it almost never crashes with the Selector and now it crashed two times in a short period of time. Seems a bit odd to me but maybe I'm just unlucky. Maybe the dump file has some infos regarding that problem.

HoLLy-HaCKeR, 2017-06-05:
From my own experience, NTR crashes 1/3 of the times I try to launch it. Pre-8.0 you would just get a native 3DS message saying that an error occured. With Luma 8.0 all these are replaced with the message you got. 
Are you absolutely sure that the current issue isn't the same you were having before upgrading, just with a different error message?

IngeniousDefault, 2017-06-05:
Really? It almost never crashed for me since I used the tool from Nanquitas. I use it pretty often and it only crashed like two or three times over the last six months.

AuroraWright, 2017-06-05:
" Pre-8.0 you would just get a native 3DS message saying that an error occured. With Luma 8.0 all these are replaced with the message you got." due to exception vectors, not a bug

closed 2017-06-05 by AuroraWright

Kaisogen, 2017-06-05:
Some apps have to crash to return to HM. Can you disable this for certain apps with say, a config?

Jan200101, 2017-06-05:
These apps dont crash they close themselfs.
You can see that when you use BootNTR Selector