LumaTeam/Luma3DS #522: It is extremely difficult to change the menu combo in Rosalina

shinji257, 2017-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (After payloads)

PIN lock: (No)

New 3DS CPU: (Default)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

When attempting to change the menu combo it doesn't wait until you have an opportunity to actually press all buttons you want in the combo.  Instead it seems to accept it the moment any button is pressed.

During testing I was (at most) able to get 2 buttons to register at once for it to take the majority of the time.  While typing this I manage to get 3 by pure luck.

sora10pls, 2017-06-05:
#491

shinji257, 2017-06-05:
Haha.  This is what I get for forgetting to review the closed issues list.  >.<

Good to know it is on the todo list.

closed 2017-06-05 by shinji257