LumaTeam/Luma3DS #524: How to patch the games?

acelerador25, 2017-06-05:
I need the steps to patch games in the new 8.0 because I can't play the FE Fates DLC (usa console/eur game/dlc) and this is in general in all games that need to be patched

Thelostrune, 2017-06-05:
There's an option for "Enable game patching" in the Luma configuration menu.
However, I'm experiencing crashes that might or might not be related to game patching as a whole. Just saying.

acelerador25, 2017-06-05:
@Thelostrune, people  I talk (I'm included) say that is the first this that do, then go to any game and don't see the patch, in this case the eur DLC from Fates

Gudf, 2017-06-05:
Did you read the wiki ?

acelerador25, 2017-06-05:
And you see that is not modified, or I'm wrong @Gudf?

Gudf, 2017-06-05:
And ? Why would this mean it's false ? As far as I know, the part about game patching is still correct, and the changes are mentioned in the release notes anyway.

acelerador25, 2017-06-05:
# @Gudf, I believe in this case that there is something different in it and we need to see that, and not even see it in the wiki as I suppose it is not updated

Red1812, 2017-06-05:
I have the same issue. If I activate game patching LayerFS and locale work but FEF cannot read the DLCs. If I deactivate game patching FEF can access the DLCs but I loose locale and LayerFS.
The issue is in Luma 8.0.

jonirrings, 2017-06-06:
I had the same issue.
jp console and jp game/dlc.
then modify the locale txt to `JPN JP JP 01` solve my problem.
maybe a new spec on language emulation is needed on the wiki.

Red1812, 2017-06-06:
I have a EUR console EUR game/dlc from eShop, not cart. I tried EUR EN and EUR EN EN 01 and the game still says SD card inaccessible.

Red1812, 2017-06-06:
I made a new test. I just remove the locale.txt file and keep the romfs. Everything works fine.
So I guess there is something wrong with the locale or we're missing something obvious...

jonirrings, 2017-06-06:
@Red1812 yes, we miss the `language emulation` spec, with which we can set the appropriate locale text

Red1812, 2017-06-06:
@jonirrings What do you mean?

jonirrings, 2017-06-06:
@Red1812 I mean, a detailed spec is need on what to write in a locale.txt

ITZYOBOYkoko, 2017-06-06:
I'm playing Fire Emblem Echoes DLC and it's not working. I enable game patching in luma config and sdcard>titles>game id>locale.txt>USA EN and still the DLC is not working. 

Did I do something wrong or is it a bug or idk lol?

Tried the language patching and stuff in bootNTR and now everything is working just fine.

drzombee, 2017-06-06:
have the same prblem.  in luma 7.1  i was patched dlc in luma menu, FE FATS run the DLC but monster hunter GEN dont run and crash. now in luma 8.0 i was patched  the dlc and FE FATES dont read the sd card and MH GEN don run. thanks for read my bad english
EL.PSY.CONGROO.

TuxSH, 2017-06-06:
Fixed by the latest commit which you can find below.
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1055982/boot.zip)

closed 2017-06-06 by TuxSH

drzombee, 2017-06-07:
TUxSH , tx bro but dont work 4 me

ITZYOBOYkoko, 2017-06-07:
@TuxSH that boot.zip is working. Can play FF Echoes DLC now. Wow.

henrysimon, 2017-09-05:
confirm working, weird though my original was 8.1, copy to usb the boot.firm become 8.0, but when i test it's working in awakening (enable patch and set the region is needed)