LumaTeam/Luma3DS #526: FBI lingers as a Zombie Title

NWPlayer123, 2017-06-05:
**System model:**

Old 3DS XL

**SysNAND version:**

11.4.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap v1.2 stable

**Luma3DS version:**

Luma3DS v8.0 stable

**Luma3DS configuration/options:**

Screen brightness: ( 1 )

Enable game patching: ( X )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( X )

--

**Explanation of the issue:**

FBI doesn't close properly and lingers as a Zombie title and prevents my system from properly shutting down, is there any way to work around that? As above, I'm on 11.4U o3DS XL, I've been able to replicate it with the 2.4.8 released CIA, both injected into H&S with GodMode9 and standalone. I used to be able to use the exception handlers to shut it down without having to hold the power button cause it'd error and "press any button to shutdown" but v8.0 doesn't have that option anymore (and it created a dump file every time which was irritating)

**Steps to reproduce:**

1. Install FBI 2.4.8 standalone or inject into H&S via GodMode9

2. "Close" it with power button or START button at the main menu

3. Try to power off the system

Kaisogen, 2017-06-05:
I am not able to reproduce this.

System Information:

Model: N3DSXL Black, US

FW: Luma 11.4

Config Settings:

>Default Emunand: 1
>Screen Brightness: 4
>Splash: Off
>PIN Lock: Off
> New 3DS CPU: Clock + L2
> Autoboot Emunand: Off
> Use EmuNand FIRM if booting with R: Off
> Enable loading external FIRMs and Modules: Off
> Enable Game Patching: On
> Show NAND or User String in System Settings: On
>Show GBA Boot Screen in patched AGB_FIRM: On
>Patch ARM9 Access: Off
>Set Developer UNITINFO: Off

1. Update FBI

2. Shut Down

3. Open FBI (Latest Version Now)

4. Close using "START" button

5. Enter poweroff screen by pressing the power button.

6. Press "Power Off"

Shuts down perfectly fine fo me.

Jan200101, 2017-06-05:
I am also not able to reproduce it but I remember having it too

NWPlayer123, 2017-06-05:
Here's a video of it in action:
https://twitter.com/NWPlayer123/status/871824151803158528

blackwing182, 2017-06-05:
@NWPlayer123 cant see the video, dont know why says that im blocked  but i  dont know you 😂

Jan200101, 2017-06-05:
I was able to reproduce it now.
I has to do with Health & Saftey

Jan200101, 2017-06-05:
After more investigation I have been able to find out that starting FBI again will not take over the lingering FBI instead it will start a new thread which will then again turn into another Zombie

Jan200101, 2017-06-05:
Other stupid things I found:
 

- a installed FBI cia softlocks if FBI is lingering
- It doesnt seem to be affecing all FBI versions according to Tux (or maybe I just misunderstood him idk)
- starting the injected FBI 5 times makes it softlock
- other apps seem to be fine with FBI lingering around

TuxSH, 2017-06-05:
I don't have the issue with my 1.4 FBI inject from the rxtools days for some reason...

blackwing182, 2017-06-05:
Are these imjection metods the same?

TuxSH, 2017-06-05:
Probably not.

This issue is probably due to injection itself. Anyways, injecting FBI into H&S is a thing of the past.

Follow: https://3ds.guide/finalizing-setup

closed 2017-06-05 by TuxSH

NWPlayer123, 2017-06-06:
in case anyone finds this, using the guide above ^, booting into homebrew launcher, loading FBI as a .3dsx and then installing the CIA version fixes it and it shuts down fine again