LumaTeam/Luma3DS #528: System crash after closing

ScorelessPine, 2017-06-05:
**System model:** N3DSXL

**SysNAND version (+emu/redNAND version if applicable):** 11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

**Luma3DS version:** 8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (Off )

PIN lock: (Off )

New 3DS CPU: (Off )
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: (x )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
After opening and closing a .cia version of the microSD Management app from the home menu, I receive a system crash and a dump file. This has only started happening since updating to Luma 8.0, and worked perfectly well on 7.1 stable. Note, this does not happen while using the microSD Management app from within the Settings app.

**Steps to reproduce:**

1.Open microSD Management app gotten from _that iso site_ (not sure if namedropping is allowed on github but just to be safe) with title ID: 0004000020024100, product code: CTR-N-HAJE

2. Close app using button on bottom right of screen and wait.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1053098/crash_dump_00000000.zip)

Margen67, 2017-06-05:
That only occurs with the unofficial cia. It's unrelated to Luma.

ScorelessPine, 2017-06-05:
Is there any way to make it stop crashing? It worked perfectly fine before version 8.0. I just don't want it to generate a 3kb dump file every time i use the app.

floppyD, 2017-06-05:
You will not get this solved, TuxSH already made his mind about forcing the handler.
[https://github.com/AuroraWright/Luma3DS/issues/525](url)

ScorelessPine, 2017-06-05:
Okay, that stinks a bit. Is there at least a way to disable the dump files? If not, I'll be done and close the issue.

TuxSH, 2017-06-05:
No.

**microSD Management .cia app is NOT made to be ran from Home Menu AT ALL**.

Please use the latest ftpd commit instead. On Windows you have the option to set up a network drive.

closed 2017-06-05 by TuxSH

ScorelessPine, 2017-06-05:
FTPD is a lot slower and I have been unable to set up a network drive, which is the only reason why I started to use this app. There is absolutely no way to prevent the crash dump log after a crash? At this point that's the only problem I have because I don't want my sd card to be clogged with unnecessary crash dumps.

piratesephiroth, 2017-06-06:
¨Yeah FTPD is garbage. MicroSD Management is the best app for transfering files between the 3DS and the PC.
It's not mean to run from the home menu simply because it tries to relaunch it's parent app and fails because it's already running. You can bypass the error by launching it from FBI's title list.

I just don't see any reason to update for this Luma 8.0 yet since most of it's rew features are great for developers only useless to the player.

james-d-elliott, 2017-06-06:
If you dislike ftpd for any reason and you're just not going to use it then use microSD Management from MSET (System Settings), it comes with every console. You may also try the FBI suggestion. Alternatively nothing is stopping you from developing something on your own or contributing to ftpd to make it suit your needs. It will take motivation to learn, sure. 

Also if you have such large files to transfer that ftpd is "slow" then it would be faster to unscrew the back cover, remove the SD card, put it into a card reader, and transfer that way. Anything over about 100MB will be faster this way regardless of what app you use due to the speed limitations of 802.1b/g.

urherenow, 2017-06-07:
No longer a viable option. For that to work with Windows 10, you must change settings that are necessary to mitigate the wannacry virus/worm/whatever it is/was

ScorelessPine, 2017-06-07:
**Please understand I have since stopped asking for an official fix for this.** I am only asking now if there is instead any way of stopping the dumping of a crash log every time it happens. **Again, I'm not asking for a fix for the crash.**
If there is no way of doing this, I may as well revert back to Luma 7.1, as I have no use for any of the other features of 8.0 at the moment.

bakugo, 2017-06-07:
@ScorelessPine
Honestly, don't bother. Just downgrade and forget about it.
Luma isn't about stability and user friendliness anymore, it's TuxSH's personal playground now and he already ruined it.

jonirrings, 2017-06-07:
@bakugo I think Luma 8 is a huge update but not ready to open release, maybe call it `beta`.
Luma 8 introduce many features, it is building a concrete base for future hacks.
naming it `luma 8 beta` maybe avoid these tremendous issues these days.
I agree that, if you require stability, just downgrade to 7.1. 
let us the `tester` test the features first, and then when it comes to be stable, enjoy the stability.

in addition, github is not a forum, it is for programmers, and it is the programmers' playground, and the same goes to 3DS hacking (Luma is hacking too), it is the hackers' playground. all what we none-hackers can do, is to enjoy and thank their hacking work along with some positive feedback.

no offense to you, but hacking is always the hackers' playground.

ScorelessPine, 2017-06-07:
@jonirrings Why, though, would they intentionally remove the ability to toggle between the system's built-in soft-crash and this forced hard crash? There is no reason that I can think of that would warrant it. Sure, call it the hackers' playground, but they release 8.0 as this big new amazing thing, when you can do just as much, if not more, on 7.1 with NTR.

jonirrings, 2017-06-07:
@ScorelessPine you know, TuxSH is young (19yrs old) and he is a hacker, as both a youngster and a hackers, he is supposed to be radical on tech and hacking.
but they may choose the too aggresive method like forcing hard crash, this feature is for debug only or feedback only I think. it make all users including the noobs to be the testers.
wish next time they can choose a conservative way for release and a radical way for beta/dev

ScorelessPine, 2017-06-07:
@jonirrings What, you mean like official builds and nightlies? **OH WAIT** they already have them separated and they still choose to make the official build users the testers.

ScorelessPine, 2017-06-07:
And to add on to this, since when was TuxSH in charge of Luma instead of AuroraWright?

jonirrings, 2017-06-07:
@ScorelessPine take a closer look. both TuxSH and AuroraWright are not in charge of the `official builds and nightlies` you said, it is `astronautlevel2`  who run script to watch this repo's commit change and build it automatically. the nightly for every watched commit, and release for every release on this repo's `release` page.
if you had use github for team works, you know a repo has a feature to add `Collaborators`, AuroraWright is still the repo owner, and in charge of it, TuxSH is a collaborator of this repo.

piratesephiroth, 2017-06-07:
Nope, @urherenow .
As far as I know the vulnerability existed in the SMB Server, which is not required for the connection to the 3DS and even if it was the case, a **fix was released in march**.

bakugo, 2017-06-07:
@jonirrings I would've already downgraded if it wasn't for the complete mess that is B9S compatibility. B9S 1.2 does not support Luma <8.0 and there's no official guide on how to downgrade (spoiler: devs don't want you to downgrade).
I'm tired of being forced into these things. First came B9S 1.0 and you HAD to update from A9LH, which was fine because the corresponding Luma build was very stable, but now with 1.2 the Luma build is a complete rushed mess so there's no excuse anymore.

ScorelessPine, 2017-06-07:
Im with @bakugo on this, I'm an end user and I don't want to be treated like a beta tester. I can't find anywhere how to downgrade from b9s 1.2 and Luma 8.0 to b9s 1.0 and luma 7.1. Why, now that they've become essentially the only viable CFW on 3ds anymore, does it seem like they've decided "fuck the users, we do what we want."?

piratesephiroth, 2017-06-07:
First of all you guys should not be so quick to update.
You saw they added a ton of features so it's obvious it comes with 10 tons of bugs.

ScorelessPine, 2017-06-07:
Its kind of hard not to when they discontinue support of what we were using beforehand...

jonirrings, 2017-06-07:
@bakugo @ScorelessPine 
`SafeB9SInstaller@0.0.7` install `b9s@1.0`, then replace `luma@8.0.0 boot.firm` with the `luma@7.1.0 boot.firm`
can't this downgrade?
I'm not native eng speaker, it `viable` means `available`?

bakugo, 2017-06-07:
@ScorelessPine
>Why, now that they've become essentially the only viable CFW on 3ds anymore, does it seem like they've decided "fuck the users, we do what we want."?  

You've answered your own question.
Why would you appeal to your users when they have no choice but to use your software anyway?
The whole B9S debacle has completely alienated every other CFW (corbenik straight up died because of it) so TuxSH and friends are free to do whatever they want without worrying about anyone going "I don't want to use this CFW anymore, I'm going to use another one"

ScorelessPine, 2017-06-07:
> I'm not native eng speaker, it viable means available?
@jonirrings I meant viable as in it is the only one that makes sense to use now since the other ones are outdated.
@bakugo I just think its bullshit that they completely change the way they act toward their userbase now that they are the only developers in the large-scale cfw community.

jonirrings, 2017-06-07:
@bakugo 
this is github, and the CFW is opened source, which means everyone can get a copy of the code of this CFW and other CFW, only if you want to build it and add the feature you need, or make a super CFW with all awesome features in other CFW.
it is github, it is supposed that all the users can code, (though not hack).
@ScorelessPine 
in fact, we could suggest in a professional programmer's courtesy, not the forum-style debate. or even more, join them, and change it ourselves.

urherenow, 2017-06-07:
@piratesephedoth you must enable netbios over tcpip for it to work on Win 10. What do you think that does? Lol...

ScorelessPine, 2017-06-07:
@jonirrings I would happily join and fix what I think needs fixing, but I have absolutely no idea how any of it works, nor do I have the time to figure it out. I just wish that they would stop treating all of us as beta testers instead of what we are: end users. There are some things that they have done in 8.0, like removing  the option for the default 3ds exception handler, that I don't see a point to. It makes many things more complicated and causes more distress than it does help the user.

jonirrings, 2017-06-07:
[3dbrew](https://3dbrew.org/) serves some info on how it works. the problem is time.
for now as a end user, just downgrade to luma 7.1 and b9s 1.0.
they are young and talent, just wish they could fix the mess and be great hackers in their promising future.

piratesephiroth, 2017-06-07:
@urherenow That's a completely different issue though.

TuxSH, 2017-06-07:
Why do issues issue like this always turn up being spamming by entitled people.

Locking.