LumaTeam/Luma3DS #529: Changing the Rosalina menu button combo only allows one button AND doesn't persist after reboot

Aluminite, 2017-06-05:
**Explanation of the issue:**

Changing the Rosalina menu button combo only allows one button

**Steps to reproduce:**

1. Enter Rosalina menu

2. Change button combination under misc, attempt to use multiple buttons

As soon as you press the first button, it says `Successfully changed the menu combo.` without letting you press more than one.

acelerador25, 2017-06-05:
In this case it's necessary to press all the buttons at the same time, then, after the reboot, is true that the code is not permanent, and is necessary for the future

sora10pls, 2017-06-05:
That's... not a bug?

Refer to #489.

pillows2, 2017-06-06:
You can use multiple buttons but you have to press them all at once.

TuxSH, 2017-06-06:
"only allows one button". This was fixed in the latest commits, see here: 
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1056541/boot.zip)

I'll do the other part.

JerryWester, 2017-06-06:
I'm not fantastic at understanding the processes behind Luma, but I suggest you write the combination to a .bin in the /luma folder and read from it at boot and after it's written to.

K0r0kS33D, 2017-06-07:
@TuxSH Is saving the hb. app application being worked on as well?

Shadow322, 2017-06-07:
What? "Saving the homebrew application application being worked on as well?"

K0r0kS33D, 2017-06-07:
@Shadow322 Saving the app that hb. app was injected into to a file on the SD card, that way we can have a designated, quick loading homebrew launcher

acelerador25, 2017-06-07:
Ok, I try the new code sistem of Rosalina, and is ok for me (press the buttons 1 by 1), but can any please can make this code wearable after the reboot?

acelerador25, 2017-06-07:
@Margen67 I say this: When you enter Rosalina for the first time (in this case after power-up) you put the oficial code, then you change the code, after the game play, you shut down the console, but if you power-up again, the code you put before it doesn't work and again need to change the code. In this case I ask that they can change the code permanently if I do a reboot in the 3ds

WubTheGame, 2017-06-07:
from the looks of some recent happenings, i'm starting to wonder whether or not Luma3DS is slowly losing some of its "noob-friendlyness"

Example: Rosalina. 'nuff said.

mysterypaint, 2017-06-07:
What if Rosalina Menu's button config read from somewhere in the luma folder as a .txt? It could solve this issue, as well as the issue for people with broken buttons.

WubTheGame, 2017-06-07:
good idea, i feel that should be implemented

Hikari-chin, 2017-06-16:
I referenced these issues [here](https://github.com/AuroraWright/Luma3DS/issues/619). Going to close this now. Thank you.

closed 2017-06-16 by Hikari-chin