LumaTeam/Luma3DS #530: NTR and Luma Game Patching issues

Sonlen1414, 2017-06-05:
It should be an issue across all models and firmware versions.
8.0 Stable using B9S 1.2
Game Patching Enabled is the only option that effected this.
So upon launching a game with Luma Game Patching setup AND having a LayeredFS.plg still in the folder from the NTR patching method, the game crashes with an exception. Removing the .plg from the NTR Patching method OR disabling Luma Game Patching fixed the issue and the game boots normally. Not sure how much of an issue, but since it could be easily overlooked, I figured an issue would be worth opening.

RubberBrony, 2017-06-06:
This has occurred with the No Lines .code patch in Pokemon Sun for me.

TLNiles, 2017-06-06:
This issue existed even before, just then LayeredFS patching was implemented. I met with that problem when I got 2 IDENTICAL patches for MHXX running at the same time via Luma and NTRboot.
Which is probably totally unnessesary, since it's the same thing, and you can make just one of CFWs run it. What's even the point of running LayeredFS patch from 2 instances at the same time, especially if it's the same one?
What I think is - those 2 memory patches, running from different instances, are simply trying to overrun each other, thus crashing the system. By description it looks like both LayeredFS and NTR patches are probably the same ones, so as I mentioned before, running both at the same time are actually unnessesary.
I'm not sure if crashing can be fixed, since I'm no dev, but I expect that it can be avoided only by creating specific checks, where Luma would overlook of there's any other patch trying to run. But it also depends on what of patcher runs earlier.
Yet again, what you do looks more like creating a mess than actual bug, since it can be clearly avoided by not making a mess of patches running from instances all over the place.

SaltyToffee, 2017-06-10:
Had a similar problem trying to randomize my Pokémon Moon, I had randomized it before but decided to stop. Once I had tried to do it again after updating to B9S 1.2 and Luma 8.0 I kept getting an exception when the game is loaded up with game patching enabled.

Hikari-chin, 2017-06-19:
Could you test if this issue occurs if you use another LayeredFS implementation like OnionFS instead of Luma's game patching? If so, it's very likely an issue with NTR, rather than Luma.

Hikari-chin, 2017-06-21:
Closing this for now; issuer has not responded to requests for further testing and this is likely to be an NTR issue. Will reopen if something proves otherwise.

closed 2017-06-21 by Hikari-chin