LumaTeam/Luma3DS #532: Larger RAM games don't load after updating B9S & Luma (SSB, MH4, Pokemon Sun, MHGen)

baabon, 2017-06-05:
<b>System Model:</b> Old3DSXL

<b>SysNAND version;</b> 11.4.0-37U SysNAND

<b>Entrypoint:</b> Boot9Strap 1.2

<b>Luma3DS version:</b> 8.0 (latest release)<hr>

<b>Luma3DS configuration/options:</b>

Default EmuNAND: ( 1 )
Screen brightness: ( 4 )
Splash: ( Off )
PIN lock: ( Off )

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Use custom path: ( )
Enable game patching: ( X )
Show NAND or user string in System Settings: ( )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )<hr>

<b>Explanation of the issue:</b>

After updating B9S 1.2 + Luma3DS 8.0 <i>(from B9S 1.0 & Luma3DS 7.1)</i> games that require to work on <i>Larger RAM Mode</i> like Monster Hunter or Super Smash Bros. haven't loaded anymore.
After shows 3ds logo the screen stays black and I must to hard reset the system.

I tried to get back B9S 1.0 + Luma3DS 7.1 <i>(when all my games worked fine)</i>, but the issue still persists.

<b>Note:</b> After I updated B9S and Luma3DS, I ran homebrew launcher <i>(with the new option of Luma)</i> and I executed tikSweep <i>(to clean all unused tickets)</i> and next cthulhu <i>(to clear my play time, step and software history)</i>.<hr>

<b>Steps to reproduce:</b>

1. Open a <i>Larger RAM</i> game (Super Smash Bros.)
2. System shows a black screen and I can't go to Home Menu pressing Home Button.
3. Try to Hard Reset or Power Off.
<br>

TuxSH, 2017-06-06:
Can't reproduce.

digidude512, 2017-06-06:
[crash_dump_00000008.zip](https://github.com/AuroraWright/Luma3DS/files/1053594/crash_dump_00000008.zip)
**System Model:** N3DSXL

**SysNAND version:** 11.2.0-35E SysNAND

**Entrypoint:** Boot9strap 1.2

**Luma3DS version: 8.0** (latest release)

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )
Screen brightness: ( 4 )
Splash: ( Before )
PIN lock: ( Off )
New 3DS CPU: ( Clock+L2 ) (Also applies to off)

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( X )
Show NAND or user string in System Settings: ( X )
Show GBA boot screen in patched AGB_FIRM: ( X )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

**Explanation of the issue:**

After updating B9S 1.2 + Luma3DS 8.0 (from B9S 1.0 & Luma3DS 7.1) much like @baabon  games that require to work on Larger RAM Mode like Pokémon Sun and Moon After shows 3DS logo I get
**An exception occurred**
Processor: ARM11 (core 0)
Exception type: data abort
Current process: niji_loc (0004000000164800)

Dump Attached.
[crash_dump_00000008.zip](https://github.com/AuroraWright/Luma3DS/files/1053593/crash_dump_00000008.zip)

xelrix, 2017-06-06:
I cant reproduce these either.
Similar setup albeit on an o3ds.
Playing MHXX just fine. With and without translation patch (patched through ntr). With and without ntr.
No luma patching though.

digidude512, 2017-06-06:
@xelrix Do you have Pokémon Sun or Moon? I can ZIP my No Line and see if that has the same problem?

xelrix, 2017-06-06:
No I dont. You could try disable game patching in luma menu and try reloading the game.

baabon, 2017-06-06:
In my Old3DS I tried disabling the game patching from Luma Menu, but the issue still persists. I uninstalled Smash Bros and installed again (different backup) and it's the same.

gnkz, 2017-06-06:
I don't know if this is related but

**System Model:** N3DSXL

**SysNAND version:** 11.4.0-35E SysNAND

**Entrypoint:** Boot9strap 1.2

**Luma3DS version:** 8.0 (latest release)

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )
Screen brightness: ( 4 )
Splash: ( Off )
PIN lock: ( Off )
New 3DS CPU: ( Off )

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: ( X )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

**Explanation of the issue:**

Pokemon moon backup crash after nintendo 3ds logo. I don't have any other game to test at the moment.

[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/1055665/crash_dump_00000002.zip)
[image]
[image]

TLNiles, 2017-06-06:
2DS, Game patching on, MHXX and Pokemon Moon runs normally. Also running latest b9s and Luma 8.0 stable.
Might be some sort of console issue, just no idea what kind of it.

srwgin, 2017-06-07:
do you change region？if changed you should system format or you can not play larger ram game

acelerador25, 2017-06-07:
I try to play the Smash cartidge in my old and I don't have any problems. In this moment I try to play my Revelations and Sun cartridge, my Moon and Meta Gear cias and then post the results

baabon, 2017-06-07:
I never did a <i>region change</i> with my O3DSXL. Yesterday, I did a <b>9.2.0 CTRTransfer</b> <i>(always of my region USA)</i>, never happened. Reinstalling the game don't work. Import the seed neither.

james-d-elliott, 2017-06-07:
Did you use a clean CTR Transfer bin from the guide?

baabon, 2017-06-07:
@james-d-elliott Yes

JackAttack2, 2017-06-07:
i have this problem too, SuMo get stuck on black screen, no game patches.

JackAttack2, 2017-06-07:
i get this dump file
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1059562/crash_dump_00000000.zip)

srwgin, 2017-06-08:
backup nand and nintendo3DS folder first
then try system format after that see what happen

baabon, 2017-06-08:
An user in another forum tried a System Format but it doesn't works. There is a similar issue when you do a Region Change.

I'll try using a friend's Pokemon Moon cartridge tomorrow, so I'll tell you about what happened.

srwgin, 2017-06-08:
I have 2 region change emunand after system format Pokémon moon worked

baabon, 2017-06-09:
Finally, I did a System Format and the issue has been solved.

closed 2017-06-13 by TuxSH

andamirorulez, 2017-06-22:
Hello, i cant boot smash 4
New 3ds, luma 8 + b9s 1.2 , 11.4.0-37U SysNAND
It stuck in 3ds boot logo, any fix for this or how to solve it?

andamirorulez, 2017-06-22:
The only option i have selected, is show "sys" in configuration