LumaTeam/Luma3DS #535: Luma 8.0 bricks o3ds

August712, 2017-06-06:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

Luma 7.1

**Luma3DS configuration/options:**

_All default except:_

Splash: ( x - before )

Autoboot EmuNAND: ( x )

--

**Explanation of the issue:**

Installed 8.0 on my o3ds, and my 3ds would not turn on. Reverted to 7.1, copied 7.1's .firm to nand, everything works. (Lemme tell you, did not enjoy the shock.)

**Steps to reproduce:**

1. Install 8.0 using installer

2. Reboot

**Dump file:**

N/A

ihaveamac, 2017-06-06:
you must follow https://3ds.guide/updating-b9s to use 8.0

August712, 2017-06-06:
Thanks. Suggestion:
Make it a bit more obvious for people who didn't know, like me.

closed 2017-06-06 by August712

sora10pls, 2017-06-06:
Or, alternatively:

Learn to read
[image]