LumaTeam/Luma3DS #537: Arm11 Crash (details inside)

Storm-Eagle20, 2017-06-06:
I'd like to report this as I've gotten a crash dump and I hope that no one else would experience this crash in the future. I would like to report this in specific as I got a dump file, which would be useful to you guys.

What happened:

1. I injected JKSM into H&S (I couldn't access the app with the HBL and I didn't want to get flagged using the .CIA).
2. I launched JKSM and backed up my game saves.
3. I decide to use Rosalina to open the HBL using Face Raiders. 
4. Upon entering Face Raiders (**BEFORE** switching the hb. title to the current application), the 3DS hangs and crashes on the splash screen before it boots.
5. Shows the crash screen below.

[Top screen of crash.](http://i.imgur.com/pTLLvjk.jpg) [Bottom screen of crash.](http://i.imgur.com/W9T1Oni.jpg)

I also have a [crash dump](https://drive.google.com/file/d/0BzA9cL8TBC9aNWZpbEdXN3NodWc/view?usp=sharing) as specified in the pictures.

Also I never had the exception handlers option enabled. This crash was really random. If any of the developers have any idea what caused it, can you leave a note here?

TuxSH, 2017-06-06:
Possible duplicate of #526.

Also stop injecting.

>  I didn't want to get flagged using the .CIA

Disable the "Sending system information" SpotPass option.

> Also I never had the exception handlers option enabled

It's now always on. Anyways this wouldn't have changed anything.

closed 2017-06-06 by TuxSH

Storm-Eagle20, 2017-06-06:
I'd like to apologize if my issue is a duplicate. I don't mean to waste your time at all. JKSM wasn't supported in .3dsx form by Rosalina, which was why I injected it to begin with.

TuxSH, 2017-06-06:
> I'd like to apologize if my issue is a duplicate.

No need to apologize, I'm just closing issues quickly.