LumaTeam/Luma3DS #538: Makefile Assumes Working Environment Uses GNU Utils

Joomla12, 2017-06-06:
So the makefile assumes one is using an environment with GNU specific utilities, which is problematic for those running a BSD variant or OS X (which uses BSD utilities). Having these conditions hardcoded in the makefile results in an incomplete compilation on anything other than Linux or Cygwin. It'd be useful if the makefile queried $OSTYPE to determine which commands to pass depending on the host OS. Of course those not using GNU/Linux or Cygwin can install GNU utilities through various means, but that's rather impractical.

urherenow, 2017-06-06:
wut

It's built with devkitArm. Same as any other homebrew.

closed 2017-06-06 by TuxSH

Joomla12, 2017-06-06:
If you'd read, I said GNU utilities. The makefile uses sed and du, and only passes arguments to them that work with the GNU variants of them.

kitling, 2017-06-25:
Last night, smartperson and I came up with a replacement for the usage of du in this script, because they were also having issues, due to being a mac user without the GNU varients of the utilities installed. `$(shell du -b $(dir_build)/modules.bin | cut -f1)` can be replaced with `$(shell wc -c < $(dir_build)/modules.bin | tr -d "[:space:]") with both the GNU varients and the BSD varients.

I do not currently have a suggestion for the replacement of the usage of `sed -e 'expression'`, though it would be useful to know the motivation of using `sed -e` there rather than just `sed 'expression'`. 

I would like to note that while this is an issue with external software, the real issue is that Luma isn't using these commands in a portable fashion. It's really annoying that one has to deal with these portability concerns, but it's unfortunately a fact of life when dealing with scripts that you want to be able to run on a wide variety of platforms.