LumaTeam/Luma3DS #539: Opening Roselina menu while in System Settings corrupts the graphics

deadair3210, 2017-06-06:
**System model:**
N3DS XL
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37U SysNAND
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.0
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )
1
Screen brightness: ( )
4
Splash: ( )
Off
PIN lock: ( )
Off
New 3DS CPU: ( )

Off
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )
X
Show NAND or user string in System Settings: ( )
X
Show GBA boot screen in patched AGB_FIRM: ( )
X
Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Every once in a while, if you open the Roselina menu in System Settings, the graphics will become corrupted and will not be fixed until leaving the Settings or rebooting

**Steps to reproduce:**

1.Open System Settings

2.Open. Roselina Menu

**Dump file:**

TuxSH, 2017-06-06:
Duplicate of multiple other issues, will fix that later after more urgent issues.

closed 2017-06-06 by TuxSH

Hikari-chin, 2017-06-25:
This should be fixed in commit 2363817.