LumaTeam/Luma3DS #540: Arm11 Crash after exiting DS wifi connections

Zabhahs, 2017-06-06:
**System model:**N3DSXL

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**11.4.0-37U Sysnand

**Entrypoint (How/what you're using to boot Luma3DS):**Boot9Strap 1.2

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**8.0 Stable

**Luma3DS configuration/options:**

Default EmuNAND: (default )

Screen brightness: (default )

Splash: ( after payload)

PIN lock: (default )

New 3DS CPU: (804 )

--

Autoboot EmuNAND: (no )

Use EmuNAND FIRM if booting with R: (no )

Enable loading external FIRMs and modules: (no )

Use custom path: no( )

Enable game patching: ( no)

Show NAND or user string in System Settings: (yes)

Show GBA boot screen in patched AGB_FIRM: (yes )

Patch ARM9 access: (no )

Set developer UNITINFO: ( no)

--

**Explanation of the issue:**
Crash after exiting DS Wifi connections settings after failing to connect to 2 wifis

**Steps to reproduce:**

1.Enter settings, go to internet settings

2.Enter 3DS Connections

3. Exit 3ds connections

4. Open ds connections

5. close ds connections

6. arm11 crash

Zabhahs, 2017-06-06:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1053909/crash_dump_00000000.zip)

TuxSH, 2017-06-06:
Fixed by latest commit.

closed 2017-06-06 by TuxSH