LumaTeam/Luma3DS #541: [BUG] Luma 8.0 region emulation function is broken

CaiMiao, 2017-06-06:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0 pre-release

**Luma3DS configuration/options:**

New 3DS CPU: ( x )

Enable game patching: ( x )

**Explanation of the issue:**

I created the locale.txt for title 000400000008B500 (Mario Kart 7 Simplified Chinese by iQue) as
`CHN ZH CN 01`
and after starting the game, both screen turned pure red and freezed. Long press shutdown is needed.
I think this problem will also reproduce on other region version of MK7.

Also I heard my friend's Rhythm Heaven Megamix (US) can't run anymore on his Japan console as region emulation is broken.

One more thing, the Luma wiki should update region emulation section, to give more detailed option for region emulation. (e.g. state/province code)

CaiMiao, 2017-06-06:
[image]

Jan200101, 2017-06-06:
Able to reproduce

TuxSH, 2017-06-06:
Possibly missing chinese font, maybe potentially missing chinese data with MK7?

What if you try with EUR FR (for example)?

> One more thing, the Luma wiki should update region emulation section, to give more detailed option for region emulation. (e.g. state/province code)

Sure, later.

retrofank, 2017-06-06:
Yes, a better detailed explantion on the Wiki would help on this please.  See this thread here:

http://gbatemp.net/threads/updating-to-b9s-2-0-luma-8-0-has-broken-language-region-on-some-games.473258/#post-7371356

Red1812, 2017-06-06:
From my testing I found out that if you have a locale.txt for a game it will not be able to access the SD card to read the DLCs. RomFS and language emulation still work but games can't access the SD card for the DLCs. If you remove the locale.txt, RomFS still works and DLCs are available again.
Hope that helps.

CaiMiao, 2017-06-06:
No, the game will load font from game itself, that's not the problem.

The only reason I would try out the new region emulation is, I can change the country information to China which is really shiny as I didn't got an iQue console.

TuxSH, 2017-06-06:
All titles are given SD access now.

Red1812, 2017-06-06:
I tried EUR EN, EUR EN EN 01, EUR EN UK 01 and EUR EN UK in the locale.txt. The games does start in English but when I try to access DLCs it says it can't find the SD Card.
If I remove locale.txt, the game stats in my console language and the DLCs work normally. If you want me to test something else I'll be happy to help.

TuxSH, 2017-06-06:
Try latest commit please.
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1055982/boot.zip)

Red1812, 2017-06-06:
Yes that works. I tried with EUR EN in locale.txt. The game launches in English, it has access to the DLCs and the shop and RomFS works too.
Do you want me to try something else?

TuxSH, 2017-06-06:
Hmm idk. Probably other regressions if you see them.

CaiMiao, 2017-06-06:
Sorry but region emu still freezes (but no more red screens) the MK7, as i tried `CHN ZH CN 01` and `CHN ZH CN`.
Only language emulation `CHN ZH` is ok but lack of region emulation.

CaiMiao, 2017-06-06:
I also tried Pokemon Moon (title `0004000000175E00`), with `CHN ZH CN 01` also freezes at loading.

MK7 freezes on logo and Pokemon Moon freezes on black screen.
[image]

Red1812, 2017-06-06:
I tried EUR EN EN 01, no freeze, DLCs are working. 
I tried EUR EN UK 01, same which is strange because UK is not in the list of countries if I read correctly patcher.c
@CaiMiao Have you tried CHN ZH ZH 01?

CaiMiao, 2017-06-06:
@Red1812 tried just now, freezes.

Red1812, 2017-06-06:
Do you have another Chinese game you could try this on?
@TuxSH Are the country and state ID optionals in locale.txt? I assumed they are since it's working for me but that might not be the intended behaviour.

CaiMiao, 2017-06-06:
@Red1812 Yes the country and state ID optional, I tested.
But I'm preferring country & state/province emulation working then I can have my country flag showing.

Pokemon Sun & Moon is an global title which includes Chinese. A good example.

CaiMiao, 2017-06-06:
http://gbatemp.net/threads/updating-to-b9s-2-0-luma-8-0-has-broken-language-region-on-some-games.473258/#post-7373732
This guy's Rhythm Heaven is still not working.

And I tested `EUR EN UK 01` or `EUR EN EN 01` on Pokemon Moon still freezes.

TuxSH, 2017-06-06:
try `EUR EN GB 01`

CaiMiao, 2017-06-06:
@TuxSH still freezes Moon.

Red1812, 2017-06-06:
I tried EUR EN GB and EUR EN GB 01 on Mario Kart 7 (eshop EUR) and it freezes with red screens.
It tried EUR EN GB on FE Fates and it worked normally. I guess the game doesn't have anything linked to the country.
So it seems the country part of locale.txt is not working properly.

AuroraWright, 2017-06-06:
For people having issues with the latest commit @TuxSH TuxSH posted:
1) is it a regression? (did it work on 7.1?). If unsure, this is a 7.1 build compatible with B9S 1.2: http://www114.zippyshare.com/v/Qt9zQPHC/file.html .
2) Are you trying to use a Japanese/American/European game on a different region (Chinese, Taiwanese, Korean) console or vice-versa? If so, note that Chinese/Korean/Taiwanese consoles have each a different font (the font is the same in European/Japanese/American ones) so that could cause the crashes.

TuxSH, 2017-06-06:
@CaiMiao I can reproduce but it shouldn't happen... weird.

What if you test w/o any save file? (ie new game prompt) ?

CaiMiao, 2017-06-06:
@TuxSH I wiped my MK7 save with JKSM and starting the game also freezes at logo.

CaiMiao, 2017-06-06:
There should nothing to do with the system font, as PKMN Moon and MK7 loads custom font from the game itself.

Red1812, 2017-06-06:
@AuroraWright It freezes only when you try to use a different country than the one the console is setup and if the game uses that parameter. 
I tried to launch MK7 bought on EUR eShop on my EUR console. The game only freezes when I use EUR EN GB or EUR EN GB 01. It works with EUR EN.
With FE Fates it works with all combinaisons.

CaiMiao, 2017-06-06:
@Red1812 No, there's nothing to do with console region.
I tried `CHN ZH JP 01` and `JPN JP JP 01` also freezes logo.

Red1812, 2017-06-06:
Yeah... The region setting on my console is FR. I just tried EUR EN FR 01 and EUR EN FR and it freezes.
With EUR EN it works but that kinda defeat the purpose.

TuxSH, 2017-06-06:
Try the latest commit please:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1056537/boot.zip)

CaiMiao, 2017-06-06:
@TuxSH It works!!!
Finally the last piece.
Thank you for the work!!!

Another reminder, a suggestion, any possibility to edit locale emulation settings directly via Rosalia?
[image]

TuxSH, 2017-06-07:
Dunno, seems like a chore :P

closed 2017-06-07 by TuxSH

Red1812, 2017-06-07:
Sorry I'm late to the party but I confirm the latest commit worked for me too.

Jimmy-Z, 2017-06-11:
Is there a list of country and state code?