LumaTeam/Luma3DS #543: [Suggestion] Cheat Searcher/NTR Plugin Loader

Monotek18, 2017-06-06:
Since a lot of time, GW have an advanced cheat searcher in his in-game menu.

So I am wondering if Luma would have that one day in his ingame menu.

Also the menu take most of NTR feature, but there is no way to use NTR Plugin... yet 
That would be nice to see that.

Margen67, 2017-06-06:
Cheat search seems possible but probably won't be added to Luma anytime soon.

Loading NTR plugins would be difficult if not impossible without the source code.

Monotek18, 2017-06-06:
Well even if that take many time it does not matter, as long as Luma will have it one day.
Abiout the plugin loading, you can RE NTR to look how it load plugin.

Margen67, 2017-06-06:
>RE NTR

That's easier said than done.