LumaTeam/Luma3DS #545: [BUG] Game patching is broken for Sega 3D Classics Collection (USA) in Luma 8.0

Epicbeardman, 2017-06-06:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

I have the USA version of Sega 3D Classics Collection installed on my european New 3DS and, due to this region mismatch, game patching is needed in order to load and save replay data. Without game patching enabled in Luma 8.0, the base game will launch but I get the following error when attempting to launch any of the games in the collection: _'Some features cannot be used because no SD card was detected_'.

However, with game patching enabled the game will freeze on a white screen after the 'Presented by SEGA' and 'M2 logos' have appeared, forcing me to close the game by pressing HOME + X.

The locale.txt file is located in `luma/titles/0004000000185e00`, with the content `USA EN`. This setting enabled game patching to work successfully in Luma 7.0 for this game, and I _think_ it was still working in 7.1, but I'm not 100% sure about that.

Interestingly enough, game patching still works fine for Sega 3D Fukkoku Archives 1 (Japan) and Sega 3D Fukkoku Archives 3 (Japan) in Luma 8.0.

TuxSH, 2017-06-06:
Fixed by the latest commit which you can find below.
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1055982/boot.zip)

closed 2017-06-06 by TuxSH

Epicbeardman, 2017-06-06:
I can confirm that the issue is now fixed. Thanks!

chashodges, 2017-07-08:
Having the same issue with newest Luma (as of 8th July '17) but with the Japanese release (Sega Fukkoku Archives 1). Same error message though.