LumaTeam/Luma3DS #546: [BUG] Game patching is broken for 3D Super Hang-On (USA) in Luma 8.0

Epicbeardman, 2017-06-06:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

I have the USA version of 3D Super Hang-On installed on my european New 3DS and, due to this region mismatch, game patching is needed in order to load and save replay data. When game patching is disabled in Luma 8.0, I am still able to play the game but the aforementioned features are not available. However, with game patching enabled the game will freeze on a white screen after the 'Presented by SEGA' and 'M2' logos have appeared, forcing me to close the game by pressing HOME + X.

The locale.txt file is located in `luma/titles/00040000000c0d00`, with the content `USA EN`. This setting enabled game patching to work successfully in Luma 7.0 for this game, and I _think_ it was still working in 7.1, but I'm not 100% sure about that.

urherenow, 2017-06-06:
Seriously, learn to read the docs before posting issues.

Gudf, 2017-06-06:
Try changing the content of locale.txt by `USA EN EN 01`

Epicbeardman, 2017-06-06:
Thanks for the suggestion. Unfortunately, the issue still persists.

urherenow, 2017-06-06:
Nope. Github isn't a support forum. If you can't figure out the docs, ask in a forum. But forget the FR. Try EN twice. Otherwise, reading the other MULTIPLE issues already posted for the same thing should tell you that it's a bug and the devs are aware of it.

urherenow, 2017-06-06:
And considering your original post has the WRONG contents for 8.0. No. You didn't tread the docs.

urherenow, 2017-06-06:
Try USA EN US 01

And look here for ideas if it doesn't work: https://github.com/AuroraWright/Luma3DS/releases

Epicbeardman, 2017-06-06:
I'm not sure how my post has the wrong contents when all I did was fill in the placeholder info, but ok.

I've tried `USA EN US 01` as per your suggestion, and also `USA EN NA 01`, but neither worked. I'm sure this is a regression in 8.0 as opposed to the additional country code or state ID being wrong, because both Sega 3D Fukkoku Archives 1 and 3 work with a locale.txt that has contents `JPN JP`.

TuxSH, 2017-06-06:
Please try the latest commit which you can find below.
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1055982/boot.zip).

It should have been fixed, I'll reopen if not.

closed 2017-06-06 by TuxSH

Epicbeardman, 2017-06-06:
I can confirm that the issue is now fixed. Thanks!