LumaTeam/Luma3DS #550: Unrespoonsive Luma3ds Configuration and Rosalina Menu

MDornH03, 2017-06-06:
Whenever i do L+Down+Select, the rosalina pops up but I can't move up, down, or press A. The only button that works is B to get out of the menu. Same thing with Luma3ds Config, except I have to shut down my 3ds to exit it. No one else has been able to help, thanks 👍.

11.4.0 37U 03ds
I followed the guide on 3ds.guide.

TuxSH, 2017-06-06:
Are you sure your buttons even work to begin with?

TuxSH, 2017-06-06:
There's no reason other than a hardware issue (buttons not properly "connected"?) for it not to work.

closed 2017-06-06 by TuxSH

Margen67, 2017-06-06:
Try this: https://github.com/AuroraWright/Luma3DS/files/1055982/boot.zip

MDornH03, 2017-06-06:
thanks so much dude, it works now

MDornH03, 2017-06-06:
nevermind, worked for a second. stopped again