LumaTeam/Luma3DS #551: I can't load a code.bin which is edited because of a new file in romfs

FlameBeam, 2017-06-07:
Sorry for my bad English.
I am a Chinses player. As you know, a number of player in China play 3DS games that is translated by some translate group like ACG.
But many times, we must edit the code.bin because we need add some files in the romfs,like font(you know the Chinese font file is different from English and Japanese).
If I use a code.bin that is edited because of some new files in the romfs, the game will be not work.
Is there any way to use this code.bin?

Hikari-chin, 2017-06-08:
I don't believe this is an error with Luma. In the code.bin, you likely need to edit it to make use of the new files.

FlameBeam, 2017-06-08:
https://mega.nz/#!dcYjnYaA!_ToNltgAMqMtkP2IjNCFnfbDKZebD00sGda17gu7bA0
Here is the patch of the Zero Escape: Zero Time Dilemma in Chinese. Because we need a Chinese input in the game, we add the "hanhua" and "system/desision/input" folders in romfs and edited the code.bin. As a cia, it works. But as a luma game patch, it dead. 
If I delete the code.bin, it works.
It's only one game in many code edited Chinese games.

FlameBeam, 2017-06-08:
I think maybe it because of the edited exh.bin?

ericjwg, 2017-06-09:
did you used a decompressed version code.bin?

FlameBeam, 2017-06-09:
I tested.
the original code.bin can work but the edited code.bin can't. 
I asked the author who edited the code, he told me because the exh.bin was edited.

ericjwg, 2017-06-09:
@FlameBeam weird. are u using the latest commit or stable release version?

FlameBeam, 2017-06-09:
@ericjwg Of course, I use the Luma 8.0.

Hikari-chin, 2017-06-19:
@FlameBeam, could you try this with OnionFS or another LayeredFS implementation and let us know if it works or has the same issue?

Hikari-chin, 2017-06-21:
Closing this, original issuer has not responded to requests for further testing. Will reopen if that occurs and this ends up being an issue specific to Luma.

FlameBeam, 2017-06-22:
I'm sorry I have test OnionFS until today. I have final exam these days.
I test the OnionFS today, it can not work.
I find the reason is that we have a costum exheader.bin, so can Luma patch a costum exheader.bin for games?

Hikari-chin, 2017-06-22:
No worries. Thank you for testing. I'll look into this further and let you know.

FlameBeam, 2017-06-22:
I tested another game(HyruleWarriors)，it's also a game with a costum code.bin and costum exheader.bin.
But it works!!! Then I try  Zero Escape again, it also works. Maybe something worng when the first test.

FlameBeam, 2017-06-22:
↑↑↑
 I mean test in OnionFS, they're working but dead in Luma.

FlameBeam, 2017-06-22:
OnionFS:
custom code.bin + custom romfs = work!
(exh.bin not changed)custom code.bin + custom romfs + new files in romfs = work!
(exh.bin changed)custom code.bin + custom romfs + new files in romfs = work!

Luma:
custom code.bin + custom romfs = work!
(exh.bin not changed)custom code.bin + custom romfs + new files in romfs = work!
(exh.bin changed)custom code.bin + custom romfs + new files in romfs = bomb!

FlameBeam, 2017-08-08:
I tested the custom code.bin with custom exheader.bin in last version(8.1.1.7). 
It still can't be run. 
Here are the dump files.
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1207500/dumps.zip)
I tested third times.

Shadow322, 2017-08-08:
Um v8.1.1.7 does not exist

FlameBeam, 2017-08-08:
Oh well, it '8.1.1.7z' ,lol

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH