LumaTeam/Luma3DS #553: Compiling error???

Shadow322, 2017-06-07:
make[1]: Leaving directory `/c/Luna3DS_Build/Luma3DS/sysmodules/loader'
make[1]: Entering directory `/c/Luna3DS_Build/Luma3DS/sysmodules/rosalina'
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -g -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -fomit-frame-pointer -ffunction-sections -fdata-sections -I/c/devkitPro/libctru/include -Iinclude -DARM11 -D_3DS   -c -o build/3dsx.o source/3dsx.c
source/3dsx.c: In function 'Ldr_CodesetFrom3dsx':
source/3dsx.c:247:26: error: 'RUNFLAG_APTCHAINLOAD' undeclared (first use in this function)
         pst->runFlags |= RUNFLAG_APTCHAINLOAD;
                          ^~~~~~~~~~~~~~~~~~~~
source/3dsx.c:247:26: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/3dsx.o] Error 1
make[1]: Leaving directory `/c/Luna3DS_Build/Luma3DS/sysmodules/rosalina'
make: *** [build/rosalina.cxi] Error 2

I don't know what i am missing

urherenow, 2017-06-07:
Belongs in a forum, not github. This is not a bug. Msg me in temo and we can have a conversation.

olefb, 2017-06-07:
You need to re-compile ctrulib. It was implemented in [this commit](https://github.com/smealum/ctrulib/commit/1ed97e8ba28fc277d863f0c2e5eebb23e65e1a3d).

Shadow322, 2017-06-07:
yea having problems compiling latest ctrulib :(

urherenow, 2017-06-07:
Re-clone it. Otherwise, impossible if using devkitArm. DevkitArm should be either first or right after devkitPro\msys in your path.

Shadow322, 2017-06-07:
ok i got ctrulib compiled and added i dont know how to install firmtool

james-d-elliott, 2017-06-07:
Read the README.md on the firmtool repo. 👍 https://github.com/TuxSH/firmtool

If you have any other questions about compilation (or anything that is not a bug) please visit one of the following links (github is a place to post bugs, not get support):

https://gbatemp.net/forums/3ds-homebrew-development-and-emulators.275/
https://discord.gg/C29hYvh

Jerry-Shaw, 2017-06-07:
Sorry to disturb, actually when I compiled the version after 8.0 under Windows 10, all seemed to be good, but the firm file leads to black screen with blue light on. It didn't happen when before 8.0. Anyone met this?

urherenow, 2017-06-07:
Did you upgrade b9s?

Shadow322, 2017-06-07:
there is newer then 1.2 or b9s?

urherenow, 2017-06-07:
@shadow, I wasn't replying to you. And I'm done here. This isn't a support forum Every ine of you need to get out and use a forum because you are all clueless.

AuroraWright, 2017-06-07:
It seems there's some issue with Windows, need to compile on Linux for now

urherenow, 2017-06-07:
Any issues between linux and Windows is usually taken care of by using msys instead of cmd. Just sayin'

Jerry-Shaw, 2017-06-07:
@urherenow Have you tried?

Jerry-Shaw, 2017-06-07:
@AuroraWright Menu is showing, and selectable, but cannot load.

Jerry-Shaw, 2017-06-07:
Tried to compile again under windows, and found that, it only failed when booting into EmuNAND.

Jerry-Shaw, 2017-06-07:
@urherenow Tried using msys, SysNAND can boot, but EmuNAND failed. Seems like Linux would be the best way.

ericjwg, 2017-06-07:
WSL for the win?

ericjwg, 2017-06-07:
@Jerry-Shaw msys build is same as cmd here. weird...

Jerry-Shaw, 2017-06-07:
@ericjwg Most the same, yes.

TuxSH, 2017-06-07:
There's a bug in luma affecting msys, leading to black screens.

To answer OP's question: update ctrulib.

Jerry-Shaw, 2017-06-07:
I compiled it right after updating ctrulib, but made no difference. Almost check all, except makerom binary.

ericjwg, 2017-06-07:
......
compile the v8.0 release commit.
still the same black screen

TuxSH, 2017-06-07:
I know there's a bug on msys just leave me time to fix it.

closed 2017-06-07 by TuxSH

Jerry-Shaw, 2017-06-07:
Haha, it's OK, just take your time. It won't be difficult for us to move to linux.
Thanks so much!