LumaTeam/Luma3DS #554: Seems Weird

knight-ryu12, 2017-06-07:
**System model:**
N3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37J SYSNAND
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

All version since Lumav7.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Before)

PIN lock: (None)

New 3DS CPU: (L2+CLK)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
When Launching Kirby Hunters Z, with B9S luma, it fails to boot and gives crashlog

**Steps to reproduce:**

1. Launch game

2.

**Dump file:**

[crash_dump_00000012.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1057869/crash_dump_00000012.dmp.zip)

knight-ryu12, 2017-06-07:
(Sorry, Including Your latest commit.)

NateDogg1232, 2017-06-07:
Did this game work with the clock+L2 cache options enabled in earlier versions?

knight-ryu12, 2017-06-07:
Yes.
I think before B9S

NateDogg1232, 2017-06-07:
Have you tried it without that option enabled? I don't have an N3DS, but I know this option causes some issues in some games.

knight-ryu12, 2017-06-07:
No Luck.

TuxSH, 2017-06-07:
In loader: Module: FS (17) | Description: NCCH hash-check failed? (fs) (391) | Summary: Wrong argument (8) | Level: Permanent (27)

Reinstall the game more properly maybe?

closed 2017-06-07 by TuxSH