LumaTeam/Luma3DS #555: [Rosalina] Add support for launching applications under target titles

Link2738, 2017-06-07:
Can this be a feature or will we have to use a different exploit?

fincs, 2017-06-07:
Duplicate of #547 and fincs/new-hbmenu#4

This is intentional. It was decided that target title selection is not useful anymore since 3dsx applications have full ARM11 system permissions now, and it is possible to directly access the savefiles of any title by using the appropriate FSUSER commands. In the meantime, you can use the CIA versions of savegame managers.

Also, HANS is obsoleted by Luma3DS's built-in ExeFS/RomFS (LayeredFS) game patching feature.

closed 2017-06-07 by fincs

Link2738, 2017-06-07:
I wasn't asking about save files I was asking about apps like braindump to dump game assets

TuxSH, 2017-06-07:
Same thing.

Link2738, 2017-06-07:
Like models,textures,ect. I know there are savemanagers galore on gbatemp
(I'll get an old 3ds and dump them on that)

Znethix, 2017-07-25:
just delete the .xml file.

I ran braindump and it worked but it doesnt let me choose the program.  Help

and what do i have to do?

fincs, 2017-07-25:
Use GodMode9 instead of braindump for dumping game assets.