LumaTeam/Luma3DS #556: Major issue in the README.md file

zayzu58, 2017-06-07:
The latest version, 8.0 has brought us some new amazing features. Unfortunately, it has also led to a very serious problem.
Back when AuReiNand and The Guide were still new, this CFW was advertised as being "noob-friendly", and it still is. That's everything Luma has had going for compared to the competition according to the people who used it and suggested to use it. Some people debated other CFWs were just as user friendly as ARN/Luma was, but that's besides the point.
Due to more recent developments, however, right now Luma is the only currently maintained CFW. That, of course, leads to problems like this: https://github.com/AuroraWright/Luma3DS/issues/528
Because of this, the README should stop describing Luma as "Noob-proof", and start calling it "Developer oriented" instead.

Jan200101, 2017-06-07:
Luma is still Noob-proof.
The issues like in #528 are a mixture of user error and the fact that Luma 8 is new

zayzu58, 2017-06-07:
That problem isn't a user error, it's a consequence of a developer feature being forced on normal users, making it developer oriented.

Jan200101, 2017-06-07:
It indeed is.
The "developer feature" is something used internaly from Luma to run.
Tux mentioned that in another Issue but dont quote me on that as I am too busy to find it in the mess that Issues has become

zayzu58, 2017-06-07:
Making a developer oriented CFW is great, but it shouldn't be suggested and referred to as something meant for normal users.

Jan200101, 2017-06-07:
How is Luma developer oriented ?
Just because it shows a error code which makes it easier to report bugs rather then saying
>Doing X will crash
You have probably seen the end of #528 so please dont start it again

zayzu58, 2017-06-07:
This feature meant for developers is creating problems for normal users. That's why it's developer oriented.

floppyD, 2017-06-07:
Like I suggested on #525 an option to turn the new exception handle off would issues like users on #528 are having, and I really don't "think" it would be that hard to re-implement it.

floppyD, 2017-06-07:
Like I suggested on #525 an option to turn the new exception handle off would solve issues like users on #528 are having, and I really don't "think" it would be that hard to re-implement it.

Jan200101, 2017-06-07:
on #528 you even got a explanation

>No, not possible, it's used by GDB [...]

Stary2001, 2017-06-07:
The correct way to solve this problem is to patch microsd mangement to _not crash home menu in the first place_, instead of simply swallowing the error and restarting home menu silently.

floppyD, 2017-06-07:
It "can" by used, right? How is GDB forcing it?

Jan200101, 2017-06-07:
@Stary2001 its rather that the people crashing with it used a CIA of it from some 3th party site

Jan200101, 2017-06-07:
@floppyD GDB is using the error handler not forcing it

zayzu58, 2017-06-07:
I get that you don't want to do anything about the other issue, it's an understandable choice. This issue is about the README file containing outdated info.

Jan200101, 2017-06-07:
The thing is
Just because Luma now has a error handler doesnt instantly make it a 'developer' cfw.
Its still the same noob friendly Luma

zayzu58, 2017-06-07:
It's causing problems to normal users who don't care about features meant for developers.

floppyD, 2017-06-07:
@Jan200101 
So if it's only using, not forcing, an option exactly like we had on previous versions could be implemented. Got my point?

Jan200101, 2017-06-07:
How is it causing problems
Ones that go above being slightly annoyed because you have to restart your system

Jan200101, 2017-06-07:
@floppyD Lets say the error handler was your CPU fan.
It will work for some time but it will just cause more problems

zayzu58, 2017-06-07:
So, let me get this straight, what you're saying is that users being annoyed to get unnecessary screens and system restarts thrown at them isn't a problem simply because you said so? Take a few seconds to think about what you typed before sending your comments, please.

floppyD, 2017-06-07:
@Jan200101 
I don't think you got it. I'm not asking to remove the handle, just to put an option to disable it. And no, that's not how it works, that analogy doesn't make sense since not a single problem will happen on my side, the regular user, by letting it turned off.

dark-samus, 2017-06-07:
This is a really really stupid issue. The problem is not Luma, as pointed out. The PROBLEM is that these applications are CRASHING home menu. In the past, this "wasn't" an issue because home menu would silently restart. Now these things are being caught by Luma's improved error handler. The solution, as usual, is to properly make your applications so that they don't crash the god damn home menu or something else!

Jan200101, 2017-06-07:
@zayzu58  Its not a problem because its not hindering you from using your 3ds at all.

All you to right now is act entitled to have a mandatory feature removed because it adds a few seconds to rebooting if something crashes.

Incase you didnt know many 3ds errors also restarted the system so you pretty much gained and lost nothing.

Instead of telling me to think before writing you should do

Hikari-chin, 2017-06-07:
The problem here is with the hacked up version of microSD Management you've installed. It literally crashes the Home Menu on exit. This isn't Luma's fault; if you want it "fixed", fix microSD Managememt yourself. Luma should absolutely not ignore crashes in any respect; allowing silent termination leads to people reporting vague "My system crashed!!" problems instead of being able to provide a crash dump, error code, the offending title ID, etc. 

Enabling the exception handlers is the best "noob friendly" option you could ask for; it allows them to report issues properly and allows said issues to be fixed quickly, as well as giving the user a general idea of what is going wrong. The device should not be a black box that no one but the developers understands; it leads to this exact for of entitlement and victim blaming as we've seen here and in prior issues.

zayzu58, 2017-06-07:
It does hinder the usage of the 3DS, as it creates a problem. And, aside from the fact that wanting the problem to be solved isn't acting entitled, as I have stated multiple times, I'm not asking for it to be changed or removed, but I'm asking to change the erroneous information contained in the README.

Jan200101, 2017-06-07:
You are asking for a change because something got added that makes errors show a a actual error and you want it removed so you have to wait 5 seconds longer.

Is it me or does not that sound entitled

Hikari-chin, 2017-06-07:
The "problem" always existed due to microSD Management not being meant to run from the Home Menu. This isn't the 1950s anymore; brushing problems under the rug for the sake of maintaining a perfect outward appearance doesn't fly. As I've stated before, the exception handlers make Luma even more "noob friendly" than without them, so the information in the README has no need to be changed.

MelonGx, 2017-06-07:
Luma Exception Handler is noob-unfriendly, indeed.

It shows a black screen to users, and makes noobs SCARED.

Although we all know that it is almost no difference between Nintendo OS's exception handling and Luma's exception handling, NOOBS DON'T.

pbanj, 2017-06-07:
Got to live talking to walls -_-
people don't understand the real issue so they blame luma. You people can always just use ftp........ Problem solved

Flubbex, 2017-06-07:
Can't Luma provide devs with a more generalized way for quitting to home somehow?

Jan200101, 2017-06-07:
@Flubbex the crash isnt a luma error.

zayzu58, 2017-06-07:
It doesn't matter that it's the intended behaviour, what matters is that a normal user who does not care about anything but using his 3DS is being disadvantaged by having something he does not care about, which does indeed make Luma "developer oriented".

TuxSH, 2017-06-07:
Closing and locking, the point has been made by @stuckpixel and @Stary2001 (and whoever else), this is not Luma's issue and will not be worked around.

closed 2017-06-07 by TuxSH

Jan200101, 2017-06-07:
👍

pbanj, 2017-06-07:
It makes it easier for people to figure out Wtf is wrong. 

Now switch to this https://mtheall.com/%7Emtheall/ftpd.cia and stop bitching