LumaTeam/Luma3DS #557: (Bug Report) Console reboot due to error with Rosalina(?)

WubTheGame, 2017-06-07:
I will do my best to provide useful information that can be used to fix the bug.
also, this bug happened before my game was able to save, so i'm quite upset.

System model:

Old 3DS XL

SysNAND version:

11.4.0-37U SysNAND (No EmuNAND)

Entrypoint:

Boot9Strap 1.2 Stable

Luma3DS version:

Luma 8.0 Stable

Luma3DS configuration/options:

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (before payload)

PIN lock: (6)

--

Autoboot EmuNAND: (off)

Use EmuNAND FIRM if booting with R: (off)

Use custom path: (off)

Enable game patching: (off)

Show NAND or user string in System Settings: (on)

Show GBA boot screen in patched AGB_FIRM: (off)

Patch ARM9 access: (off)

Set developer UNITINFO: (off)

--

**Explanation of the issue:**

I was playing Warrior's way on mii plaza, i had beat the emperor after several tries, and i decided to take screenshots using Rosalina. i pulled up the menu several times to take several pictures, but at one point i got an error when trying to take another screenshot. after pressing the home button to upload a screenshot to miiverse, i got the following error that i was able to write down:

An Error occurred (errDisp)

Error type: logged

Process ID: 103
Process name: menu
Process title ID: 0x00040003000008f02

Adress: 0x00164244
error code: 0xd860466c

Press any button to reboot

Steps to reproduce: (this might not quite work for you)

1. Take multiple Screenshots until you get an error and cannot take any more (you might not get this first error idk)

2. exit the Rosalina menu and press the home button

if it worked you should have got the same error as explained above. if it doesn't, i apologize but i can't guarantee anything.

**Dump file:**
i do not yet have a dump, though i can try to get one.

WubTheGame, 2017-06-07:
P.S (i can't locate the edit post button, the one i found only lets me edit the title) as of now i can't get a dump, but i'll be able to try within 20 minutes

AuroraWright, 2017-06-07:
Is the SD card full? (please check this)
Are you using a custom theme in the home menu?
Were you using LayeredFS for this game?

WubTheGame, 2017-06-07:
No, i have 1.7 GB left.
Yes.
No, i have game patching disabled.

WubTheGame, 2017-06-07:
also, i have a suggestion: maybe there can be some kind of "add-on/extension" system for Rosalina? maybe they could be called Galaxies and that other developers could create little bits of additional functionality. i for one think there could be something that records an action or series of actions, and then automatically repeats it, then it could be paused while the Rosalina menu is up for easy management without them getting in the way.

Stary2001, 2017-06-07:
extensions idea already suggested over at #512 - like the Galaxies name though

WubTheGame, 2017-06-07:
Thanks Stary. glad to know SOMEONE likes my ideas.

WubTheGame, 2017-06-07:
but it also could be used for stuff completely unrelated to cheats and modding.

closed 2017-06-11 by TuxSH

WubTheGame, 2017-08-03:
why is nobody trying to help solve this? i submitted a bug and it's being ignored.

deadair3210, 2017-08-03:
Its not being ignored. Its almost impossible to fix a bug that can't be reproduced reliably, and so far nobody else has come forward and said that the same thing happened to them. It could have just been a one off thing, it happens sometimes

AuroraWright, 2017-08-04:
this was fixed in recent commits, "logged" errdisp errors no longer show the error screen

WubTheGame, 2017-08-05:
@deadair3210 oh, well thanks for letting me know.
@AuroraWright How could i know?

but whatever. anyway, the problem is solved and i finally know it.

polopoi, 2017-08-28:
I have the same problem as WubTheGame for Pokémon moon bug when I confront trainers at the volcanic park : 

An error occurred (ErrDisp)

Error type: generic
Process ID: 38
Process name: niji_loo
Process title ID: 0x0004000000175e00

Adress: 0x002b37401
Error code: 0xo96045b

Press any button to reboot

please help me 

News : system 11.5.0-38E , FBI is 2.4.11,Luma 8.1

Jvolt7, 2017-10-17:
I had the same problem just now.

Old 3DS XL 
11.6.0-39U
Boot9strap 1.3
Luma 8.1.1
2.10 of a 32GB SD Card

The thing is, I believe I did something I shoudn't, earlier I set PIN LOCK option to the luma on CTR Nand and in the SD Card, then, I was playing Mario Kart when I finished a Cup and it returned to the main screen and at this point I pressed and holded the power button to turn off and it gave the same error screen and telling to press any button to reboot, I tried a couple of times going to the MK7 main scream and turning off the console and the error persisted, later I was playing Street Fighter and it gave tem same error but while I was playing a match, so I decided to reinstall everything (Luma, Godmode etc) and set Luma to CTR NAND again, apparently it fixed but i'm not sure