LumaTeam/Luma3DS #558: [SUGGESTION] Make Rosalina optional

sks316, 2017-06-07:
There's been reports of people experiencing problems with the new Rosalina system module (myself included, with a freeze in ACNL, find it in the Luma3DS GBAtemp thread) and some of us would prefer not to use Rosalina (again, myself included) so it would be amazing if there was an option in Luma Config to enable/disable Rosalina. TBH the bugs make Rosalina seem more like a beta, so I would disable it until it fully works.

sks316, 2017-06-07:
To address any questions on how The Homebrew Launcher would work when Rosalina is disabled, it would be booted from hblauncher_loader and would use the otherapp payloads.

TuxSH, 2017-06-07:
No.

closed 2017-06-07 by TuxSH

WubTheGame, 2017-06-07:
i actually like Rosalina, i think it has bugs and needs work, yes, but it also has potential.

Hikari-chin, 2017-06-07:
Rejected. See #556,