LumaTeam/Luma3DS #559: 3DS Will Not Boot After Updating Luma (11.4)

Eqouy, 2017-06-07:
My 3DS was updated to 11.4 so I was thinking I should update Luma, but after updating. I went to turn the device back on but only for a second will the blue LED turn on then it will turn off, and there in nothing on the screen. I have recently found out of I used another SD card, booted into GodMode9 then unmounted the SD card, switched back to the original one and restored from the backed up NAND (11.4) would get it working. I am fine now but I cant seem to find a way to open any payloads from Luma (GodMode9 and Decrypt9) on this single SD card.

ericjwg, 2017-06-07:
did you read the release log???
https://github.com/AuroraWright/Luma3DS/releases/latest

urherenow, 2017-06-07:
1)This isn't a support forum. This is to report legit bugs

2) You need to provide more info than that. What do you mean by updating Luma? To which version? did you install and/or update b9s? Latest Luma hourly requires latest b9s (1.2). 

Other causes:

fake or bad card. Troubleshooting guide asks to run a diagnostic tool to check, does it not?
small card with 64k cluster size
with some cards maybe 64k clusters even with 32GB or more

3ds.guide -> use it if trying the latest

closed 2017-06-07 by TuxSH