LumaTeam/Luma3DS #560: Luma3DS-84d736c ???

Shadow322, 2017-06-07:
all newer builds compiled on *nix black screen and blue led causes u to force shutdown on N3DS/XL but work fine on O3DS/XL

dogtopus, 2017-06-08:
That commit got reverted. It's probably broken.
Did you try the latest commit?

WubTheGame, 2017-06-08:
i recommend just using the stable build.

TuxSH, 2017-06-08:
Or one I've posted in the region emulation issues.

MelonGx, 2017-06-08:
Ask Linux builders to build it for you.
Current codes cannot be properly built by Windows environment.
Also do not use astro's nightly until he fix his environment.

jonirrings, 2017-06-08:
@MelonGx can it be built under *nix, like mac os?

MelonGx, 2017-06-08:
@jonirrings Dunno about Mac. We only have Windows and Linux.

jonirrings, 2017-06-08:
@MelonGx got it, thank you

WubTheGame, 2017-06-08:
oh how i wish i could go back to Windows.

ericjwg, 2017-06-08:
@WubTheGame windows build doesn't work either.
only nix-ish

WubTheGame, 2017-06-08:
regardless, i don't really like mac.

ihaveamac, 2017-06-08:
@WubTheGame please don't comment just because, only if you have something to add to the conversation.

WubTheGame, 2017-06-08:
okay..

TuxSH, 2017-06-08:
Fixed.

closed 2017-06-08 by TuxSH