LumaTeam/Luma3DS #563: Pokemon Sun Random Crashing

survive9, 2017-06-08:
My sister is playing through Pokemon Sun for the first time. I installed CFW on her 2ds and updated to boot9strap 1.2 recently along with the latest official release of Luma3ds. She is experiencing very random and unreliable crashes that happens every so often. The usually happen right after battles are finished or when entering new routes. She is on latest firmware, we also tried uninstalling the patch and playing without it to no success. She has not experienced any crashes with Super Mystery Dungeon, which is a cart while Pokemon Sun was a digital download. Here are some of the dumps for when crashes happened. 

[crash dumps 2ds.zip](https://github.com/AuroraWright/Luma3DS/files/1059802/crash.dumps.2ds.zip)

Kaisogen, 2017-06-08:
Make a backup of your SD card and test it.

Sounds like a bad SD card. Does this happen to other titles? Or is this only to Sun and Moon?

survive9, 2017-06-08:
Only to Sun/Moon. She has been playing Super Mystery Dungeon with no problems so far. Might be an issue with extended memory mode or something

Kaisogen, 2017-06-08:
I'm not a dev, so I can't understand these dumps.

But I can troubleshoot basic problems. Can you test your SD card using [H2TestW](https://3ds.guide/h2testw-(windows))? Or the appropriate tool for your OS? Are you using Luma 8.0? Nightly or Regular?

Please also put a copy of your config file here, as it may hold something we don't know here.

james-d-elliott, 2017-06-08:
Related https://github.com/AuroraWright/Luma3DS/issues/532

survive9, 2017-06-08:
Using Luma 8.0 Stable. 

[config.zip](https://github.com/AuroraWright/Luma3DS/files/1059939/config.zip)

After testing with H2TestW, the tool found no errors with the SD card. Since the SD card was formatted, we will test again to see if that somehow solved the issue.

Kaisogen, 2017-06-08:
I... Can't read the bin. It's binary. Can you just GIVE me the settings you have?

survive9, 2017-06-08:
Lol oops my b. No pin lock or splash screen. The only option checked is Enable game patching. I will test the game with the option turned off

Zarklord1, 2017-06-08:
playing pokemon moon at this moment... works perfectly fine on my n3ds xl. though this is a cart and not a digital download... i can post my settings if you want buts its pretty much standard stuff from the guide...

Hikari-chin, 2017-06-09:
This is very likely related/a duplicate of #532. @survive9, could you try formatting your system and seeing if that fixes the issue? I cannot reproduce this issue on Pokemon Moon or Sun, digital or cart.

survive9, 2017-06-09:
Yea I can try that @Hikari-chin just give me some time to get everything set back up.

Kaisogen, 2017-06-10:
Try installing an older version of Luma. If that doesn't work, it's something with your console.

survive9, 2017-06-10:
Yup we have downgraded to Luma 7.1 and boot9strap 1.0 hope this solves it

JakeK0202, 2017-06-13:
I'm having the same issue happening, but in Alpha Sapphire instead on n3ds. Crashed on luma 8.0/b9s 1.2 and same with luma 7.1. But didn't downgrade b9s to 1.0. Would this possibly fix it?

TuxSH, 2017-06-14:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1073198/boot.zip)

Try the latest commit.

TWonderchild, 2017-06-14:
Got the same on my 2ds and o3ds. But this only happens when the game has been running for a long time like 2+ hrs.

JakeK0202, 2017-06-14:
Same here. Only after it's been running for a while will it crash, but when it crashes after that point is very inconsistent. Tried the latest commit, still crashing. Not sure what else it could be at this point

Hikari-chin, 2017-06-21:
Could anyone with this issue try a later commit like [this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-1782827.zip)? I cannot reproduce this issue no matter how long I play Pokémon Sun or Moon, online or offline.

DeathAngel74, 2017-06-21:
Sun, Moon, Ruby, Y, Black all work fine. All legit carts. Even with romfs and ips patches.

TWonderchild, 2017-06-21:
I found the issue (at least on my device) it's input redirection. If I'm not using input redirection there arent any crashes. When I'm using it it will crash sometimes, sometimes my console doesnt even shut down properly when I have been using it (the game closes, blackscreen and then nothing happens) but this might be even a diffrent issue?!

TKSTScopinator, 2017-06-22:
this happened to me until i turned off game patching in the luma config and saved it after it rebooted it was fine again. And i went back to check game patching to see if it was still going to be a problem to deal with and turned it on and launched sun with it and it started working again. so try turning on if u had it off save the config launch sun if sun crashes go back to luma config and turn it off and see if it works. After this test your game patching on a different game or the same pokemon to see if it is the problem.

DeathAngel74, 2017-06-22:
Game patching isn't the issue. I'm using nova sun, Umbra moon, wilting y, eternal x, rutile ruby, and star sapphire. All romfs and ips patches with luma 8 game patching.

Hikari-chin, 2017-06-27:
Going to close this. Have had multiple people, including myself test Pokemon Sun and other extended memory games, and not be able to reproduce this. If it's found to be actual issue later, I'll reopen it.

closed 2017-06-27 by Hikari-chin