LumaTeam/Luma3DS #564: System crashes after closing microSD Management from system settings (ARM11 error)

souleater98, 2017-06-08:
[New 3DS XL]

11.4.0-37U SysNAND,

Boot9Strap 1.2

[Luma 8.0]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (Off )

PIN lock: (Off )

New 3DS CPU: (Clock )

Autoboot EmuNAND: ( no)

Use EmuNAND FIRM if booting with R: (no )

Enable loading external FIRMs and modules: (no )

Use custom path: (no )

Enable game patching: (yes )

Show NAND or user string in System Settings: ( yes)

Show GBA boot screen in patched AGB_FIRM: (yes )

Patch ARM9 access: (no )

Set developer UNITINFO: ( no)

**Explanation of the issue:**

Whenever I finish using the microSD Management from the system settings, I would proceed to press the power button to turn off my New3DSXL and this error pops up.

**Steps to reproduce:**

1. Go to microSD Management on system settings, and connect to your PC.

2.After connecting to PC, press the power button to turn of your 3DS.

**Dump file:**
[image]

MelonGx, 2017-06-08:
Can't reproduce your issue on
- Sys 11.4.0-37J
- Luma 8.0-76d7a035

MelonGx, 2017-06-08:
Reproduced your issue on
- Sys 11.4.0-37J
- Luma 8.0-76d7a035

rfenix15, 2017-06-08:
With this commit https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-6c6d710.zip the problem solve, is a luma 7.1 but work en bootstrap9 1.2 and not rosalina menu

Valeri0p, 2017-06-08:
Isn't that from the legacy branch?

DeathAngel74, 2017-06-08:
you can try this one too
[Luma3DSv8.0-a16d1eb.zip](https://github.com/AuroraWright/Luma3DS/files/1060171/Luma3DSv8.0-a16d1eb.zip)

Hikari-chin, 2017-06-08:
This issue has been posted on IRC as well. It's unknown whether this is a Luma issue or simply Luma's exception handlers catching an OS bug (like another issue recently). Either way, it has been acknowledged and will be fixed/worked around.

Valeri0p, 2017-06-08:
So _maybe_ #528 was a legit issue?

Margen67, 2017-06-08:
@Valeri0p The MSD Management app isn't meant to be ran from the home menu so that's probably a seperate bug.

Hikari-chin, 2017-06-08:
@Valeri0p No, that's a separate crash; microSD Management causes the Home Menu to svcBreak() if opened and exited from it.

Otakusensei, 2017-06-09:
@DeathAngel74 That one didn't solve the error for me.

TuxSH, 2017-06-13:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1070019/boot.zip)

Latest commit. It's a Nintendo bug anyways.

closed 2017-06-13 by TuxSH

rfenix15, 2017-06-14:
TuxSH that's not happent in luma 7.1 with this commit https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-6c6d710.zip

Hikari-chin, 2017-06-14:
No, it still happens, it just isn't caught by the inferior exception handlers.

rfenix15, 2017-06-14:
But, in 7.1 show the normal message to turn off the 3ds

rfenix15, 2017-06-14:
A last question, where i can download the latest commits compiled?