LumaTeam/Luma3DS #565: Latest commit 427a059 ???

Shadow322, 2017-06-08:
can anyone be kind and compile a copy plez

james-d-elliott, 2017-06-08:
https://astronautlevel2.github.io/Luma3DS/

Get the second last one: a16d1eb

427a059 just reverts a commit.

Shadow322, 2017-06-08:
if it was there sure

404

File not found

The site configured at this address does not contain the requested file.

If this is your site, make sure that the filename case matches the URL.
For root URLs (like http://example.com/) you must provide an index.html file.

Read the full documentation for more information about using GitHub Pages.
GitHub Status — @githubstatus

DeathAngel74, 2017-06-08:
I hope this helps!

[Luma3DSv8.0-a16d1eb.zip](https://github.com/AuroraWright/Luma3DS/files/1060102/Luma3DSv8.0-a16d1eb.zip)

james-d-elliott, 2017-06-08:
https://transfer.sh/3tvxY/boot.firm here. Just built it.

DeathAngel74, 2017-06-08:
doh! ninja'd

Shadow322, 2017-06-08:
@james-d-elliott THANKS!!!

closed 2017-06-08 by Shadow322

Hikari-chin, 2017-06-08:
This is not the proper place to ask for this. Please do not do this again.