LumaTeam/Luma3DS #566: Taiko 3 romfs not working

GoldNoway, 2017-06-08:
Hi,
I already stated that on GBAtemp, but I figured it'd be worth it to mention it here too.

When using the romfs translation patch for Taiko : Dokodon - Mystery Adventure, as a "romfs" in the correct folder, to load the game in english, it just doesn't work : Taiko is still in Japanese.
That's the case ever since romfs feature was implemented on Luma. I didn't care previously because it was bootable with Hans, but Hans doesn't show up in the new Homebrew Menu, so Taiko 3 romfs is basically not working.

All DLC work correctly though.

Titleid for this game is 0004000000190E00.

Shadow322, 2017-06-08:
U need to extract the roofs into a folder u can't use a single .roofs file

Hikari-chin, 2017-06-08:
Could you link to the GBAtemp thread in question? As @Shadow322 said, if the romfs hasn't been extracted into a folder (rather than a .romfs, romfs.bin, etc.) it will not work.

GoldNoway, 2017-06-08:
Well I just mentioned that in the Luma general thread, and that was not as well explained as here. I think I made a thread a while ago but I don't really remember that one :')

In any case, the romfs, named "romfs" without any extension, is located on luma/titles/0004000000190E00/. I don't think there's an error here, and as stated earlier that file worked previously with Hans.

Hikari-chin, 2017-06-08:
I've never used HANS before, so you'll have to clarify. Is the "romfs" here a file or a directory? If it's the former, that will not work. The romfs must be extracted into a directory called "romfs".

GoldNoway, 2017-06-08:
Oh ? The way I understood it, the file was to be named "romfs" and put in the same directory as the locale.txt file, did I make a mistake ? :w

closed 2017-06-08 by GoldNoway