LumaTeam/Luma3DS #567: Since 8.0 menuhax broken

GothicIII, 2017-06-08:
**System model:**
N3DS
**SysNAND version (+emu/redNAND version if applicable):**
SYS 9.2.0-20
Emu 11.4.0

**Entrypoint (How/what you're using to boot Luma3DS):**

b9s (emu)
menuhax (sys)

**Luma3DS version:**
stable 8.0 to current hourly

**Luma3DS configuration/options:**

Doesn't apply here.

--

**Explanation of the issue:**

Since luma 8.0 menuhax stopped working. It was the only convenient way to return to emunand after playing a gba game.

**Steps to reproduce:**

1. Load a gba game (VC injection).

2. Return to home menu. The way agb_firm works (in order to save the game) luma must load sysnand. 

3. If menuhax is installed in sysnand, luma will crash. 

**Dump file:**
(ZIP with winrar not supported)
[crash_dump_00000002.txt](https://github.com/AuroraWright/Luma3DS/files/1060745/crash_dump_00000002.txt)

Shadow322, 2017-06-08:
Update sysnand to 11.4 menuhax is not needed anymore

GothicIII, 2017-06-08:
This is not a solution. I have my reasons why I'm using emunand.

james-d-elliott, 2017-06-08:
With Luma 8.0 you can load Homebrew from Rosalina without the need for menuhax. All old hax entry points are broken.

Shadow322, 2017-06-08:
That reason being?

GothicIII, 2017-06-08:
Will this be fixed? In this state the new luma is useless for me and I need to revert back to b9sv1.

Shadow322, 2017-06-08:
Broken nand chip?

GothicIII, 2017-06-08:
The nand chip produces crc-errors on specific adresses. I needed to modify and restore the data inside the fat-partition to get the device working again, so don't want to touch sysnand.

Margen67, 2017-06-08:
#495

Hikari-chin, 2017-06-08:
Per #495, exploits requiring *hax payloads don't (and can't) work with Luma3DS 8.0. Please update to [fincs' hbmenu](https://github.com/fincs/new-hbmenu/releases) and enable "Autoboot EmuNAND" in Luma's config screen. If this still isn't satisfactory, you can try to use to the legacy branch of Luma.

AuroraWright, 2017-06-08:
"so I don't want to touch sysnand."
For what it's worth, even by booting into menuhax data is written to sysnand by the OS, so using menuhax still puts you at risk
"It was the only convenient way to return to emunand after playing a gba game."
If you reboot again while in SysNAND it will boot EmuNAND.

GothicIII, 2017-06-08:
Ok thank you. 
menuhax data is written 1 time only, so no issues here. An issue would be if I wanted to install more gba games.
I'll revert back to b9sv1, then. I don't want to explain the kids how to restart the 3ds. I assure you after a week I would need to do the unbrick again and on that unit it takes a lot of time.

closed 2017-06-08 by GothicIII