LumaTeam/Luma3DS #570: sdcard (rare) random unmounts

pcs3rd, 2017-06-08:
idk if my sdcard is the problem. Every once in a while, right off the bat, when opening the lid, the sdcard does not remount and theme and everything else disappears until I reboot.

Hikari-chin, 2017-06-08:
Could not reproduce, attempted ~100 times. Likely a hardware failure, test your SD card with [H2testw](https://3ds.guide/h2testw-(windows).html) or another tool. If the SD card comes back fine, it may the SD card slot itself; the one on the 3DS is not built for constant repeated cycling.

zennnnnnnnnnnnnnn, 2017-06-08:
This is likely a poor SD card.

pcs3rd, 2017-06-09:
ok. thanks. every once in a while, it does similar while exiting rosalina

pcs3rd, 2017-06-09:
It's actually stabalized over the last 2 hours

Hikari-chin, 2017-06-09:
Closing. If this turns out to be an actual issue, I'll reopen it at a later date.

closed 2017-06-09 by Hikari-chin