LumaTeam/Luma3DS #571: [Suggestion] Add screen brightness adjustment option into Rosalina Menu

NitroCipher, 2017-06-08:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

ahoischen, 2017-06-08:
First of all, this seems to be a duplicate of #59 which is now over one year old. The maintainers should decide on how this is handled. I also request the issue description to be trimmed since it doesn't include any content. 

The suggestion seems to be to add a backlight adjustment to the Rosalina menu. It is most likely derived from NTR's backlight control and seems useful for those times when the home menu isn't accessible. While fine-grained control is nice, I don't see it as necessary since the home menu is now available at any time that Rosalina is.

TuxSH, 2017-06-08:
Will be added.

Hikari-chin, 2017-06-16:
See the reference. Closing this now. Thank you.

closed 2017-06-16 by Hikari-chin

CookieXD, 2017-12-24:
How long will it take to add it? Can't wait anymore... (I am very impatient)

Hikari-chin, 2017-12-24:
Entitlement won't get you anywhere.

CookieXD, 2017-12-24:
I just was curious how long it might take... Guess I just have to wait.

Shadow322, 2017-12-24:
2-4 years

CookieXD, 2017-12-24:
`2-4 years` seems obvious to me.

Shadow322, 2017-12-24:
And might even take as long as 5 1/2-6 years