LumaTeam/Luma3DS #572: Screentearing with system hang upon taking screenshots in the Sun/Moon Demo.

Davixxa, 2017-06-08:
**System model:**

O3DS (Ambassador-old)

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**
8.0 Stable
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this 

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 4 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( N/A )

--

Autoboot EmuNAND: ( Off )

Use EmuNAND FIRM if booting with R: ( Off )

Enable loading external FIRMs and modules: ( Off )

Use custom path: ( N/A (Option not showing up) )

Enable game patching: ( Off )

Show NAND or user string in System Settings: ( On )

Show GBA boot screen in patched AGB_FIRM: ( On )

Patch ARM9 access: ( Off )

Set developer UNITINFO: (Off  )

--

**Explanation of the issue:**

Upon taking a screenshot using the new screenshot feature, sometimes, the screen will begin tearing. It also causes the system to hang, though the music will still be playing. I've had it happen once in the Sun/Moon Demo, after battling Team Skull, and once after exiting System Settings.

**Steps to reproduce:**

1. Boot up Sun/Moon Demo

2. Beat Team Skull

3. Open Rosalina menu.

4. Take Screenshot

**Dump file:**

TuxSH, 2017-06-08:
Happens very occasionally. Will fix that later (after other bugs).

closed 2017-06-08 by TuxSH

Hikari-chin, 2017-06-25:
This should be fixed in commit 2363817.