LumaTeam/Luma3DS #574: .CRO ROMFS Patching Crashes System

thatsmyface, 2017-06-08:
**System model:**

2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Enable game patching: (x)

**Explanation of the issue:**

The system crashes when running a modified .cro file using Luma Game Patching. (Tested on Pokemon Omega Ruby with modifying starters)

**Steps to reproduce:**

1. Place a .cro in /luma/titles/[titleid]/romfs

2. Run game (the crash happens for me immediately after the openning cut scene)

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1062213/crash_dump_00000000.zip)

AuroraWright, 2017-06-08:
Are you sure the cro is proper?

thatsmyface, 2017-06-08:
Oh my bad it looked like the software I was using put a random file in there.

closed 2017-06-08 by thatsmyface