LumaTeam/Luma3DS #575: [Suggestions] Add Power off and Reboot to the new menu and other enhancements

Red1812, 2017-06-08:
I'd like to make a few suggestions for the new shiny menu (I like it very much):
- Add Power off and Reboot options. 
- Save the selected app for booting Rosalina even after a reboot. I guess this one might be asked a lot. Keep the option to revert back to hb launcher in case you don't remember which app you chose (yes it happened to me...) or display the selected app.
- Add the option to create the titleid folder in /luma/titles while in game. That's purely for convenience so people don't have to look for the titleid before creating a locale.txt and/or romfs for the game.

TuxSH, 2017-06-08:
At least reboot, and 2). Will do.

Red1812, 2017-06-08:
That might be a stupid comment but why didn't you add the option to launch Rosalina from the menu? Does it need an app to hook to?

james-d-elliott, 2017-06-08:
Don't you mean launch homebrew/boot.3dsx?

Red1812, 2017-06-08:
Yes, I call it Rosalina because that what it says once you booted it. But you're right there is the Rosalina menu and the Rosalina homebrew menu.

trbocode, 2017-06-09:
@Red1812  My fork has Power Off and Reboot. You can use it in the meantime.

Red1812, 2017-06-09:
Unless I'm mistaken the latest release in your fork doesn't include the fix for locale.txt and DLCs, so I'll pass. But thanks.

trbocode, 2017-06-10:
@Red1812 Making a new (pre) release just for that.

Red1812, 2017-06-11:
@trbocode Nice it works great. The power off is a bit wonky though. 
I expected it to be almost immediate, so when nothing happened I tried to exit the menu and suddenly the screen became all glitchy and it turned off. 
After another try I finally got it, I have to exit the menu for the console to turn off. But the screen glitch remains, people are not going to like that :)

Stary2001, 2017-06-11:
reboot/poweroff implemented in commit 37eb21d

Hikari-chin, 2017-06-16:
See the reference. Closing this now. Thanks.

closed 2017-06-16 by Hikari-chin