LumaTeam/Luma3DS #578: Loading emuNAND is broken (4d99143 and above)

LioMajor, 2017-06-09:
**System model:** new3DS
**SysNAND version (+emu/redNAND version if applicable):** sys9.2 / emu11.2
**Entrypoint (How/what you're using to boot Luma3DS):** BS9 1.2
**Luma3DS version:** Last good [461035b](https://github.com/AuroraWright/Luma3DS/commit/461035b2127b0d404350a115cdee5ea23905ba9d)
**Luma3DS configuration/options:**

Default EmuNAND: (1)
Screen brightness: (4)
Splash: (before)
PIN lock: (no)

New 3DS CPU: (default)
Autoboot EmuNAND: (yes)
Use EmuNAND FIRM if booting with R: (no)
Enable loading external FIRMs and modules: (no)
Use custom path: (no)
Enable game patching: (no)
Show NAND or user string in System Settings: (yes)
Show GBA boot screen in patched AGB_FIRM: (no)
Patch ARM9 access: (no)
Set developer UNITINFO: (no)

--

I use debian to compile Luma3DS with latest devkitpro/libctru/firmtool/makerom.
Since [4d99143](https://github.com/AuroraWright/Luma3DS/tree/4d99143da02cc3ab4d260a03f045e005efd4bd25) booting emuNAND is broken.
[461035b](https://github.com/AuroraWright/Luma3DS/commit/461035b2127b0d404350a115cdee5ea23905ba9d) is the current last version that is capeable to boot emuNAND.

**Steps to reproduce:**

Inject emuNAND into your memory card and try booting it > blackscreen

Shadow322, 2017-06-09:
1.    Do not use hourlies who haven't upgraded devkitARM + ctrulib + firmtool to the latest.
2.    If you build Luma yourself, update your devkitARM + ctrulib + firmtool to the latest.
 3.   If everything don't work, ask a Linux builder to build it for you. Windows has bad compatibility on this        wave of building unusable Luma.

MelonGx, 2017-06-09:
N3DS SysNAND is also down at 949d219.
Only O3DS can be properly booted currently.

MelonGx, 2017-06-10:
52a1883 properly booted N3DS EmuNAND.
Don't know about O3DS EmuNAND.

LioMajor, 2017-06-10:
Can confirm, 52a1883 bootet my N3DS EmuNAND too.

Clement95, 2017-06-10:
This issue can be closed now ? Stable version of Luma3DS doesn't can
boot EmuNAND

Clement95, 2017-06-10:
Actually no, someone could give me the zip file of the hourly build 52a1883, I can't compile :/

MelonGx, 2017-06-10:
Posted it to GBATemp.

Clement95, 2017-06-10:
Doesn't work for me with N3DS.. https://filetrip.net/dl?POkglWdXg1

sora10pls, 2017-06-11:
\>EmuNAND

Margen67, 2017-06-11:
If you're not:
Region changing
Bypassing the title limit
A developer

EmuNAND is just a waste of SD card space lol

ihaveamac, 2017-06-11:
emu/rednand being useless or not does not really solve the issue. this is an issue tracker, so it not loading is a valid issue.

TuxSH, 2017-06-11:
Probably a broken commit

TuxSH, 2017-06-13:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1070018/boot.zip)

closed 2017-06-13 by TuxSH

Clement95, 2017-06-13:
My EmuNAND doesn't display a HOME Menu with the boot.firm of your
boot.zip, I have a black screen, but when I press Power, the Power
screen is displaying