LumaTeam/Luma3DS #579: Powers on and then instantly powers off after updating

NoobForeverANDever, 2017-06-09:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

nvais0n, 2017-06-09:
Did you update boot9strap to 1.2? luma3DS 8.0 requires it.

Margen67, 2017-06-09:
Nobody can help you if you don't even provide any details.

nvais0n, 2017-06-09:
@Margen67, he probably has a old boot9strap, same thing happens to me, I updating everything through 3DS.guide. Now I have luma 8 working.

NoobForeverANDever, 2017-06-09:
ill see when i get home

NoobForeverANDever, 2017-06-09:
all right Im back and I checked and turns out I didnt update boot9strap

nvais0n, 2017-06-09:
Working now?

closed 2017-06-09 by Hikari-chin