LumaTeam/Luma3DS #580: Crash in JKSM homebrew

seiya-git, 2017-06-09:
**System model:** 2DS
**SysNAND version:** 14.4.0-37E
**Entrypoint:** Boot9Strap
**Luma3DS version:** 8.0

**Luma3DS configuration/options:**

- [ ] Default EmuNAND
- [ ] Screen brightness
- [ ] Splash
- [ ] PIN lock
- [ ] Autoboot EmuNAND
- [ ] Use EmuNAND FIRM if booting with R
- [ ] Enable loading external FIRMs and modules
- [ ] Use custom path
- [x] Enable game patching
- [x] Show NAND or user string in System Settings
- [ ] Show GBA boot screen in patched AGB_FIRM: ( )
- [ ] Patch ARM9 access
- [ ] Set developer UNITINFO

**Steps to reproduce:**

1. Start Homebrew launcher 2.0
2. Start latest version JKSM from github repository with pressed L and R.
3. Select "System Titles"
4. Press and hold UP.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1064966/crash_dump_00000000.zip)

Hikari-chin, 2017-06-09:
Does JKSM crash if used as a .CIA? If not, this is very likely an issue with JKSM itself or @fincs new hbmenu.

Phalk, 2017-06-09:
I've compiled a version of JKSM that is compatible with Homebrew Menu v2 and I can't reproduce this error with it. Check JKSM thread on gbatemp..

fincs, 2017-06-09:
hbmenu 2.0 has nothing to do with how the 3dsx apps are loaded. The code to do that is in Rosalina.

Hikari-chin, 2017-06-09:
Closed. Issue with JKSM, per @Phalk's post and the GBAtemp thread.

closed 2017-06-09 by Hikari-chin