LumaTeam/Luma3DS #582: Can't install Stable build with Luma Updater 2.1.2

WubTheGame, 2017-06-09:
O3DSXL 11.4 USA B9S 1.2 Luma 8.0

whenever i try to install the latest Stable build from Luma Updater, it "Succeeds" and when i go back into it it shows a dev build next to installed version. should i just get the payload from the releases page and see if that works?

Hikari-chin, 2017-06-09:
Luma Update is not integrated into Luma, it is separate software not maintained by a different developer. If you believe there is an issue with it, please open an issue on [Luma Update's own GitHub repo](https://github.com/KunoichiZ/lumaupdate/issues).

closed 2017-06-09 by Hikari-chin

WubTheGame, 2017-06-09:
ok thanks.

Margen67, 2017-06-09:
Also `Luma updaters that don't support Boot9Strap/Sighax won't work.
This is due to support for non-B9S/Sighax entrypoints being dropped.`

WubTheGame, 2017-06-09:
i was using a B9S updater....