LumaTeam/Luma3DS #583: An Exception Occurred

Bonesaww, 2017-06-10:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.3.0-36U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

Updated B9S and Luma last night with the most recent non-dev non-ntr versions and am greeted with [this](https://i.imgur.com/w0Tb8Ng.jpg) error upon booting to SysNAND.

Chainloader/EmuNAND booting still seem to work just fine.

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/1065476/crash_dump_00000001.zip)

zennnnnnnnnnnnnnn, 2017-06-10:
Make sure everything is in the right place, and I reccomend updating to 11.4.

Kaisogen, 2017-06-10:
> updating to 11.4

Can't do that if SysNand isn't booting.

Reinstall BootStrap 1.0, and then go back to Luma 7.1.

Update to 11.4, and make sure everything is functional.

If you don't see anything out of place, reinstall 1.2

Bonesaww, 2017-06-10:
Tried it. Boots to black screen and none of the usual fixes work (No SD/Cart/Safe Mode). Chainloader/EmuNAND still work fine. Guess I'll try a 9.2 CTR transfer.

Kaisogen, 2017-06-10:
Uhh

Replace Luma with 7.1 then downgrade to Bootstrap 1.0? You don't have to CTRTransfer.

Bonesaww, 2017-06-10:
> Tried it.

Bonesaww, 2017-06-10:
Well the 9.2 CTR transfer seems to have fixed it, so I'm gonna just go ahead and close this. Still confused as to what went wrong in the first place though.

closed 2017-06-10 by Bonesaww

david-zzj, 2017-08-11:
hi, can you tell me how to reinstalling b9s 1.0 ？
now i am b9s 1.2