LumaTeam/Luma3DS #585: Crash when exiting system settings

Eliastik, 2017-06-10:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: 1(X)

Screen brightness: 4(X)

Splash: Before (X)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

My 3DS crashed after I used the Nintendo DS Internet Settings and after I exited the system settings with an ARM11 exception.

**Steps to reproduce:**

1. Use system settings app

2. Go to Internet Settings

3. Go to Nintendo DS Internet Settings

4. Exit the Nintendo DS Internet Settings

5. Touch "Back" to go to the home screen of the system settings

6. Exit the system setting with the "Exit" button.

**Dump file:**

PJBeans, 2017-06-11:
I'm no admin, but I will say this:

This "bug" was already reported [here](https://github.com/AuroraWright/Luma3DS/issues/453). To paraphrase @AuroraWright, this is normal, even without custom firmware. The reason for this is because the Home menu crashes for some reason after exiting NDS Internet Settings, followed by existing System Settings. On stock firmware, the system reboots, thus the crash is not noticeable to the user. However, in Luma, the crash gives full information of what happened.

Next time, make sure that the issue wasn't already reported. Don't worry, we all make the mistake from time to time  ;)

To fix:
Press any button to reboot and go about your day :)

TuxSH, 2017-06-11:
Duplicate

closed 2017-06-11 by TuxSH