LumaTeam/Luma3DS #586: Crash when installed boot9strap 1.2 and copy luma 8.0.7 to SD and open 3ds.

wang80919, 2017-06-10:
**System model:**
New3ds
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
V9.1.0-20J
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.0.7
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ()
1
Screen brightness: ( )
4
Splash: ( )
off
PIN lock: ( )
off
New 3DS CPU: ( )

--
off
Autoboot EmuNAND: ( )
off
Use EmuNAND FIRM if booting with R: ( )
on
Enable loading external FIRMs and modules: ( off)

Use custom path: ( )
off
Enable game patching: ( )
off
Show NAND or user string in System Settings: ( )
on
Show GBA boot screen in patched AGB_FIRM: ( )
off
Patch ARM9 access: ( )
off
Set developer UNITINFO: ( )
off
--

**Explanation of the issue:**

**Steps to reproduce:**

1.
use  Homebrew Launcher ot install boot9starp ok
2.
copy lunm 8.0.7 to sd
3.
open 3ds
4.
sysNAND loading crashed.
enuNAND 11.4.0.37J loaded ok.

**Dump file:**

Hikari-chin, 2017-06-10:
Can't reproduce this; also 8.0.7 doesn't exist, or if it does exist as a commit you did not specify it correctly.

closed 2017-06-10 by Hikari-chin

jonirrings, 2017-06-11:
Luma3DSv8.0.7z -> 8.0.7……:joy:

wang80919, 2017-06-11:
ok it is 8.0

Clement95, 2017-06-11:
Do you have updated to boot9strap 1.2 ?